PSP-Archive/ARK-4 #4: No work on PSVITA

theheroGAC, 2021-04-04:
tested and retested but ARK-4 does not work on psvita

JoseAaronLopezGarcia, 2021-04-04:
How are you installing it?
If you are using the Bubble installed by FastARK then you need to copy VitaBubble/H.BIN into ARK_01234 (replacing).
If you are using other installations of ARK you might have to copy the K.BIN file from the previous installation.

theheroGAC, 2021-04-04:
Hi thanks for the answer, I'm using Arkfast. In the Vitabubble folder inside ARK.zip there is no H.BIN

JoseAaronLopezGarcia, 2021-04-04:
Correct, I must have forgotten to update release files.
You can find it here: https://github.com/PSP-Archive/ARK-4/blob/main/loader/usploit/vitabubble/H.BIN

theheroGAC, 2021-04-04:
yes now work.Only replace H.BIN inside old a ARK_01234

theheroGAC, 2021-04-04:
for you thank you
https://github.com/theheroGAC/ArkFast/releases/tag/2.31

JoseAaronLopezGarcia, 2021-04-04:
Thanks, I will add a link to it on the README.

theheroGAC, 2021-04-04:
Just for confirmation, from the old ARK-2 with this ARK-4 just replace the H.BIN file? Thanks

JoseAaronLopezGarcia, 2021-04-04:
yes, that's where the kxploit resides in that ARK-2 release (weird that they put it there and not K.BIN).

closed 2021-04-04 by theheroGAC