PSP-Archive/ARK-4 #7: Wrong savedata title

Yoti, 2021-04-19:
Savedata has an ARK-2 title instead of ARK-4.

JoseAaronLopezGarcia, 2021-04-20:
I'm aware of this but there's no single savedata/param.sfo editor that I've tried with good results (I always end up with a "corrupted save").
I'll leave this open if anyone finds one that works.

Yoti, 2021-04-20:
You must decrypt it before editing.
https://github.com/pspdev/pspsdk/blob/master/src/utility/psputility_savedata.h#L130
What is hase game used to create a data?

JoseAaronLopezGarcia, 2021-04-20:
I'm not sure, it was made a long time ago.

Yoti, 2022-02-02:
@JoseAaronLopezGarcia tested on real PSP w/ 6.60 & 6.61 firmwares.
[ark-4_savedata_files.zip](https://github.com/PSP-Archive/ARK-4/files/7989472/ark-4_savedata_files.zip)

JoseAaronLopezGarcia, 2022-02-04:
> @JoseAaronLopezGarcia tested on real PSP w/ 6.60 & 6.61 firmwares. [ark-4_savedata_files.zip](https://github.com/PSP-Archive/ARK-4/files/7989472/ark-4_savedata_files.zip)

Awesome! Thanks a lot, will add it to the repo.

closed 2022-02-05 by Yoti