PSP-Archive/ARK-4 #10: can i switch proc and ark

Kittencore0914, 2021-11-28:


Kittencore0914, 2021-11-28:
between lme and pro，like“ProSwitcherLme”
pro and art-4？

JoseAaronLopezGarcia, 2021-11-28:
You can use Infinity bootloader to triple boot ME, PRO and ARK. There is one download for it in the releases section of this project.

JoseAaronLopezGarcia, 2021-11-28:
Latest release of ARK contains the Infinity installer.
https://github.com/PSP-Archive/ARK-4/releases/download/r91/ARK.zip

Kittencore0914, 2021-11-28:
> 您可以使用Infinity引导加载程序将ME，PRO和ARK三重引导。在此项目的"发布"部分中有一个下载。

first i install art-4 on ofw according to your tutorial，then i install proc in proc (on art-4)and select “proc" model in the Infinity.but i found art-4 disappeared in the end.
Which step i did wrong？ Can you teach me the right steps？
I really like art-4.I'm sorry for my bad English.

JoseAaronLopezGarcia, 2021-11-28:
Are you using the infinity version in the ark download?
It's the only infinity version with support for ARK, other links are usually outdated.

Kittencore0914, 2021-11-28:
> 您是否在方舟下载中使用了无限版本？它是唯一支持ARK的无限版本，其他链接通常已经过时。

yes,use yours in ”ark“zip the first time.
"2.05"

JoseAaronLopezGarcia, 2021-11-28:
Try first installing pro and then installing ark.
Then use infinity to change cfw.

Kittencore0914, 2021-11-28:
> Try first installing pro and then installing ark. Then use infinity to change cfw.

On PSP:

* For PRO Users:
- If you are running an existing PRO installation (Including Infinity).
- Copy ARK_PRO_Update to /PSP/GAME.
- Copy the folder ARK_01234 into /PSP/SAVEDATA/
- Run the updater from PRO, it will replace PRO files with ARK files.
- To go back to running PRO, simply run PRO Update to reinstall PRO CFW files.?

or
On PSP:

* For PRO Users:
- If you are running an existing PRO installation (Including Infinity).
- Copy ARK_PRO_Update to /PSP/GAME.
- Copy the folder ARK_01234 into /PSP/SAVEDATA/
- Run the updater from PRO, it will replace PRO files with ARK files.

your  tutorial means "replace"s

**Maybe need to modify the tutorial？**

JoseAaronLopezGarcia, 2021-11-28:
Ok the readme seems to be outdated.
You no longer need to use the ARK PRO Updater since Infinity is superior. The steps are simpler:
1. Install pro like you normally do.
2. Install ARK by running the Live loader (NOT the ARK PRO Updater).
3. Install the Infinity version from the ARK download.

You can ignore the ARK PRO Updater since that one is used to replace PRO with ARK, which allows ARK to run as an update to PRO rather than a standalone CFW.

Kittencore0914, 2021-11-28:
> Ok the readme seems to be outdated. You no longer need to use the ARK PRO Updater since Infinity is superior. The steps are simpler:
>
> 1. Install pro like you normally do.
> 2. Install ARK by running [...]

Queston1:
need”Copy the folder ARK_01234 into /PSP/SAVEDATA/“？
Q2：
new tutorial like this
- Install pro like you normally do.
- Copy "ARK Live" to /PSP/GAME.
- Copy the folder ARK_01234 into /PSP/SAVEDATA/
- Install ARK by running the Live loader (NOT the ARK PRO Updater)
- It will install FLASH0.ARK into flash0 and launch the CFW.
- Install the Infinity version from the "ARK"zip.
- Then switch "ARK-4"  "proc" which your want.
Right?

And "K.BIN" need move(copy or cut) any more?

JoseAaronLopezGarcia, 2021-11-28:
Yeah that seems about right, yes you need to copy the ARK_01234 into savedata folder. As for K.BIN it can be in either the savedata folder or the Live Loader folder.

Kittencore0914, 2021-11-28:
> 是的，这似乎是正确的，是的，您需要将ARK_01234复制到saveata文件夹中。至于K.BIN它可以位于saveata文件夹或Live Loader文件夹中。

your mean "K.BIN file from ARK_Live folder" "ARK_01234 folder" must be same?
the“K.bin" Need to be  ”k.bin" in ARK_Live in the zip or "K.bin" in ARK_01234 folder in the zip?
i see two "k.bin"s sha256 is diffent

JoseAaronLopezGarcia, 2021-11-28:
No they don't need to be the same, the Live Loader uses its own K.BIN (its already included in its own folder).
The K.BIN in the savedata folder is a dummy, you can replace it if you want but that is only useful for old legacy exploits on PS Vita.
Generally you don't want to replace or move files on PSP, just copy ARK_01234 to savedata folder and Live Loader to game folder, only on Vita you need to change a few stuff and depending on the setup.

Kittencore0914, 2021-11-28:
> No they don't need to be the same, the Live Loader uses its own K.BIN (its already included in its own folder). The K.BIN in the savedata folder is a dummy, you can replace it if you want but that [...]

> No they don't need to be the same, the Live Loader uses its own K.BIN (its already included in its own folder). The K.BIN in the savedata folder is a dummy, you can replace it if you want but that [...]

according to your tutorial 。ark-4 disappeared。

JoseAaronLopezGarcia, 2021-11-28:
1. Install pro as you normally would.
2. Install ARK_01234 in savedata folder and Live Loader in game folder. Launch the Live Loader to install and boot ARK.
3. Install Infinity from ARK download.

In which step does it cause trouble?
When you say ARK disappears you mean that you try to run it with Infinity and it doesn't work?

Kittencore0914, 2021-11-28:
it doesn't work。
and in ark mode and run “Infinity from ARK” ，psp will be “Blue Screen”

Kittencore0914, 2021-11-28:
> * Install pro like you normally do.
> * Copy "ARK Live" to /PSP/GAME.
> * Copy the folder ARK_01234 into /PSP/SAVEDATA/
> * Install ARK by running the Live loader (NOT the ARK PRO Updater)
> * It will install FLASH0.ARK into flash0 and launch the CFW.
> * Install the Infinity version from the "ARK"zip.

then i Install ark-4 in 6.60Infinity ofw（NO art-4 NOproc）
into art-4 is ok，run  Infinity，Blue Screen.

Kittencore0914, 2021-11-28:
do you have email? i can send the pic of blue screen.

JoseAaronLopezGarcia, 2021-11-28:
Oh yeah, Infinity doesn't seem to run when launcher from ARK (I don't know if happens with other CFW?), you should run it from OFW (hold home button when booting the psp) and launch it from there.

Kittencore0914, 2021-11-28:
hahahah... i hold home botton when booting，then into the proc...  wth
then i run Infinity to change to ark-4, then reboot...  into proc again

so funny..OTL

JoseAaronLopezGarcia, 2021-11-28:
When you hold home button it boots into PRO instead of OFW? And when you tell it to boot into ARK it boots into PRO?
This seems like a really strange issue.
When you run the Live Loader, does it install flash0 files and boots ARK correctly?

JoseAaronLopezGarcia, 2021-11-28:
Did you actually install the Infinity update?
You don't just have to install it on update folder, you also have to flash the update within the Infinity app, otherwise it won't be fully installed.

Kittencore0914, 2021-11-28:
yeah i install Infinity，and run it.

JoseAaronLopezGarcia, 2021-11-28:
One you run infinity you have to press R button to flash the update.

Kittencore0914, 2021-11-28:
> flash

i  use Chronoswitch tp keep my psp with clean 6.60.then install your  Infinity. it is not update.
it is all clean.

Kittencore0914, 2021-11-28:
> One you run infinity you have to press R button to flash the update.

you mean i must install "O“ Infinity 2.0.3,then update 2.0.5 with your Infinity？

JoseAaronLopezGarcia, 2021-11-28:
Which psp model?
Did you install ark and pro before infinity?

JoseAaronLopezGarcia, 2021-11-28:
Try this:
1. Use chronoswitch for clean 6.61 (or 6.60)
2. Install pro and ark on OFW using their live loaders.
3. Reboot into OFW again and install Infinity.

Kittencore0914, 2021-11-28:
psp3000 09G

Kittencore0914, 2021-11-28:
I'm charging,hahh debugging in the web with Developer

JoseAaronLopezGarcia, 2021-11-28:
Maybe we can talk better on discord.

Kittencore0914, 2021-11-28:
> Maybe we can talk better on discord.

OK，can you give me a address？

JoseAaronLopezGarcia, 2021-11-28:
I'm in the psp discord community.

Kittencore0914, 2021-11-28:
ilovezzg#6660

Kittencore0914, 2021-11-28:
ilovezzg#6660

acidflare, 2022-01-12:
Using a psp-2001 the ark infinity installer doesn't boot from OFW, ark needs to be installed and running first, the installer won't work if you don't hold R on the first boot though, the system will shut down if you don't hold R when attempting to launch the ark infinity installer even with ark running. The infinity installer will work as intended after all installation is finished. I'm not sure if needing to hold R is an intended function however it is not existent in the 2.0.3 version of the infinity installer nor is it included in the readme.md for this project or on the gbatemp post it wasn't until I saw the comments here that I knew I needed to hold R while starting the Ark Infinity Installer.

This was done on a system with a previous infinity setup with pro and ME installed and a CIPL flash.

- Run OFW 6.61 installer through chronoswitch.
- Install ark from ofw
- Install Infinity v2.0.3
- Shutdown system
- Boot ark live from the installer
- Start Ark Infinity Installer while holding R
- Update Infinity to v2.0.5
- Boot Ark Live from the installer once more
- Run the Ark Infinity 2.0.5 Installer
- Enable Ark in the modules section
- Reboot and Enjoy a persistent Ark installation.

acidflare, 2022-06-15:
> Just one further question here.
> Pro as a permant CFW uses cIPL - the custom boot loader to boot the actual modified FW.
> Infinity's entry point is exploit based after IPL loading - if i get anything right.
> Should we be able to install infinity (2.05) on top of a cIPL modified system? 
> This should give us the option to boot to CFW (permanent) on cold boot by pressing home?

There's no need for PRO on a system with ARK installed. Using Infinity with ARK gives the same CFW-On-Boot effect using a cIPL with PRO installed does.
Follow the steps above that I posted and you'll be all set to go.

svensvond, 2022-06-15:
I was actually asking If that ist possible. Iam ansking because i could cold Boot pro in demand then.

JoseAaronLopezGarcia, 2022-07-14:
I might have fixed the issues with Adrenaline in the latest builds. Have anyone tried it?

closed 2022-08-07 by JoseAaronLopezGarcia

Sukotto-1999, 2023-02-21:
Anyway, how can i uninstall ARK4 CFW completely? i want to revert back from ARK4 to LME2.3

JoseAaronLopezGarcia, 2023-02-21:
> How can i uninstall ARK4 CFW completely? i want to revert back from ARK4 to LME2.3. In my opinion, LME2.3 is better than ARK4 because for ARK4, i have minimalistic features in ARK4 that i can't use.

Just run the ME installer.

Sukotto-1999, 2023-02-21:
Oh, but wait. So does this mean i can install LME 2.3 while ARK4 cfw is installed. Can LME installer overwrite ARK4 CFW?

JoseAaronLopezGarcia, 2023-02-21:
> Oh, but wait. So does this mean i can install LME 2.3 while ARK4 cfw is installed. Can LME installer overwrite ARK4 CFW?

They can both co-exist at the same time and you can switch back and forth (check ARK's readme section on how to install as a Live CFW).
In any case, can you tell me which features you are missing from ME? I'd like to implement them.

Sukotto-1999, 2023-02-21:
Just ARK and LME CFW?, i will test if the PSP still runs without the memory stick.

Sukotto-1999, 2023-02-21:
Anyway, can 6 files in the installer mean installing LME 6.61 CFW overwrite ARK4?

JoseAaronLopezGarcia, 2023-02-21:
> Just ARK and LME CFW?, i will test if the PSP still runs without the memory stick.
> 
> edit: It turns out, in the System info, it still says ARK4 infinity 2.

You have to run the ME installer from OFW (hold Home button on boot).

> Anyway, can 6 files in the installer mean installing LME 6.61 CFW overwrite ARK4?

no files are overwritten, they don't have the same names so they can both be installed.

Sukotto-1999, 2023-02-21:
Oh, so there are 6 ARK4-specific modules. So, may i install LME?

JoseAaronLopezGarcia, 2023-02-21:
> Oh, so there are 6 ARK4-specific modules. So, may i install LME?

Sure. So are you gonna tell me what feature from ME is missing? Or was that just a made-up excuse? I mean I don't care if you use ME or not, I love ME myself, ARK has lots of ME stuff in it and the developer of ME was also one of the developers of ARK. But you said there's a feature in ME that you can't do in ARK and this is very important for me, can you please tell me what this feature is?

> Hopefully i can save time purchasing another PSP for UMD videos.

How is ME better than ARK for playing UMD videos? You can't change UMD drive region to play region locked UMDs in ME.

Sukotto-1999, 2023-02-21:
OH! So ARK4 is better than LME for playing region-locked UMD videos on a Japanese PSP. It's not better than LME for M33 driver.

Demonic098, 2023-02-21:
Why do you need M33 driver?

JoseAaronLopezGarcia, 2023-02-21:
> OH! So ARK4 is better than LME for playing region-locked UMD videos on a Japanese PSP. It's not better than LME for M33 driver.

So far ARK is the only CFW that can change UMD region to allow region locked UMD playback on any PSP model, mainly because this feature is rather new and hasn't been done before, so no ME is not better for UMD playback, it's actually worse!

How is the M33 Driver in ME better than Inferno 2? It's slower, supports less formats, has more bugs and consumes more RAM. Inferno 2 is much better in any meassure.

Sukotto-1999, 2023-02-21:
> How is the M33 Driver in ME better than Inferno 2? It's slower, supports less formats, has more bugs and consumes more RAM. Inferno 2 is much better in any meassure.

> Why do you need M33 driver?

First, Inferno 2 is better for ARK4 in case of M33 or Inferno having issues in LME. Second i prefer M33 driver on LME because, when using isochange.prx plugin with Sora no Kiseki SC on ARK4 with inferno 2, the system cuts to sleep mode when using a combination needed for the plugin to swap discs (both L+R triggers and start) and then it goes back on to the change disc screen, the game crashes when i come back to the Disc Change screen and attempting to press the Circle button.

Finally, it turns out LME and ARK4 can actually co-exist at the same time! No more bricks!

JoseAaronLopezGarcia, 2023-02-21:
> First, Inferno 2 is better for ARK4 in case of M33 or Inferno having issues in LME. Second i prefer M33 driver on LME because, when using isochange.prx plugin with Sora no Kiseki SC on ARK4 with [...]

This is interesting! I'll try to solve the issue. In any case, you can just open a new Issue to report this bug.

Sukotto-1999, 2023-02-21:
Well, actually you can list M33 or whatever as a non recommended driver.

JoseAaronLopezGarcia, 2023-02-21:
> Well, actually you can list M33 or whatever as a non recommended driver.

This is offtopic, but if you speak Japanese I could use some help translating this for the CFW:
https://github.com/PSP-Archive/ARK-4/blob/main/extras/xmbctrl/XMB_EN.TXT

I need an XMB_JP.TXT for ARK_01234. If you can help I will appreciate it.

Sukotto-1999, 2023-02-21:
Luckily, i can use DeepL to translate it in Japanese. So here it is for XMB_JP.TXT. This may help.

カスタムファームウェアの設定
プラグインマネージャー
USBチャージ
オーバークロック
パワーセーブ
オートブートランチャー
PSP Goの一時停止を無効にする
メモリ増設
メモリースティックの高速化
インフェルノキャッシュ
PSP Goの旧プラグインのサポート
ソニーロゴをスキップする
PIC0とPIC1の非表示

JoseAaronLopezGarcia, 2023-02-21:
> 

Thanks a lot! Added to the repo, new releases and pre-releases will now have it incorporated.

JoseAaronLopezGarcia, 2023-02-22:
> Luckily, i can use DeepL to translate it in Japanese. So here it is for XMB_JP.TXT. This may help.
> 
> カスタムファームウェアの設定 プラグインマネージャー USBチャージ オーバークロック パワーセーブ オートブートランチャー PSP Goの一時停止を無効にする メモリ増設 メモリースティックの高速化 インフェルノキャッシュ PSP Goの旧プラグインのサポート ソニーロゴをスキップする PIC0とPIC1の非表示

How would you translate this line?

Block hibernation on PSP Go

Sukotto-1999, 2023-02-22:
PSP Goのハイバネーションをブロック

JoseAaronLopezGarcia, 2023-02-22:
> PSP Goのハイバネーションをブロック

Thanks a lot!

[4 more replies]