PSP-Archive/ARK-4 #12: nothing happens

hardlevel, 2022-01-13:
Hi, I inserted the folders as said on the redme here, one folder at savegame, and other in game folders, when I try to load the launcher, a black screen with some things writen and after that it go back to the XMB, I inserted the infinity folder in game folder, and it appears as corrputed, What I missed?

I`m using a PSP 3000 and already have Infinity installed

krazynez, 2022-01-16:
Infinity EBOOT.PBP should go into ms0:PSP/GAME/UPDATE/

hardlevel, 2022-01-17:
ok, but what I have to do to see the cfw running? Nonthingh happens after the loader closes

krazynez, 2022-01-17:
After running the Ark Loader it will return you to the XMB. Press select (to see Arks VSH menu) or go i to the system information and you can see the software version is ARK 4.X.X Live

hardlevel, 2022-01-17:
I tried but it still pro infinity, nothing says ark

krazynez, 2022-01-17:
Do you mind joining the [PSP Homebrew Community Discord](https://discord.com/invite/bePrj9W) would be easier to go through all the steps. Then we can work on better understanding of what is going on so I/someone else can update the README to make it easier for future people that are having issues to avoid the confusing.

krazynez, 2022-03-17:
Did you manage to get this to work?

JoseAaronLopezGarcia, 2022-03-18:
A few things to note:

- If you run ARK loader within PRO CFW, it will not work (will load PRO). You must load ARK from OFW (hold home button when powering on).

- The infinity EBOOT.PBP must be inside ms0:/PSP/GAME/UPDATE/ folder, if you simply copy the "Infinity" folder to /PSP/GAME it won't work (says its corrupted), so just rename the "Infinity" folder to "UPDATE".

closed 2022-03-27 by JoseAaronLopezGarcia