PSP-Archive/ARK-4 #13: Not working on 3.65

KOlek0, 2022-01-13:
Get error -1 could not execute kernel function

JoseAaronLopezGarcia, 2022-01-19:
The kernel exploit ARK is using for PS Vita currently only supports 3.60, we will look into adding 3.65+ compatibility soon.

McFly75, 2022-03-28:
Maybe change the

Compatible with all PS Vita models on firmware 2.10 and up.

to

Compatible with all PS Vita models on firmware 2.10 and up to 3.63.

in the description.
(Spend so time on 3.65 before finding out I needed 3.60 in order to get Ark-4 to work)

Dimensional, 2022-06-10:
When are you going to correct the Readme to state it will not run on 3.65+? It still just says 2.10 and up, which is commonly being repeated by others. Please update it.

JoseAaronLopezGarcia, 2022-08-27:
It is compatible now:
https://github.com/PSP-Archive/ARK-4/releases/tag/r1814

Note: you can use the K.BIN file in the VitaBubble folder to load older versions of ARK too.

closed 2022-08-27 by JoseAaronLopezGarcia