PSP-Archive/ARK-4 #14: Cant install PSP Go 6.20 LME-2.3

Carbon271, 2022-01-21:
Tried installing as per the instructions but got an error (-1) when the kernel privileges in the installer were getting escalated Idk if it supports 6.20 but I prefer 6.20 on the Go since it doesn't break the Resume feature am I missing something though kind of want to give this a try is nice to see development work for the PSP still.

Carbon271, 2022-01-21:
So upon checking noticed a little late lol that it only works for 6.60 and 6.61 is there any way to get a version working for 6.20 I wouldn't mind taking the time to test and help if needed.

JoseAaronLopezGarcia, 2022-01-22:
There's no support for firmware bellow 6.60 to allow for compatibility with all devices using the same base binaries. Getting this to work on older firmwares is not planned for the time being as I continue to finish pending features and fix the last bugs. Maybe in the future.

Carbon271, 2022-01-23:
Hey thanks for the reply no worries I will try to see if I can find a work around for the PSP Go resume feature on 6.60 thanks alot for the hard work it is really appreciated.

Got it working on my Vita and PSP 1000 everything been good so far.

closed 2022-01-23 by Carbon271