PSP-Archive/ARK-4 #16: Snes9xTYme Mod 180404

McFly75, 2022-03-24:
Snes9xTYme Mod 180404 crashes when trying to launch a snes rom

McFly75, 2022-03-24:
Thought it might be because of the Media Engine, so tried Snes9xTYLcm Mod 171023, that does not use the Media Engine (as far as I know) but the PSP still crashes when trying to load a rom (works fine with Pro-C2 CFW)

krazynez, 2022-03-27:
Cannot replicated this, I have tried ` Snes9xTYme Mod 180404` and also `Snes9xTYLcm 0.4.2 Mod Rev. 28` and both are running fine from the XMB & ArkMenu (Custom Launcher) possibly you have a plugin that is causing and issue? Try to disable all plugins and test again.

McFly75, 2022-03-28:
Disabled ALL plugins and SnesxTYme Mod 180404 still crashes the PSP (screen freezes and PSP shuts down)
Tried different roms and tried on both a PSP 3000 and a PSP GO (different memory cards)

McFly75, 2022-03-28:
Tried with a New memory card and put nothing on it except Ark-4 files & SnesxTYme + some snes roms.
So no plugins or anything.
SnesxTYme starts fine, menu works fine, but when starting a rom the screen freeze and after a few seconds the PSP shut down.
(PSP 3000)

JoseAaronLopezGarcia, 2022-03-28:
Can you post a link to download the exact version you are using?

McFly75, 2022-03-28:
https://github.com/esmjanus/snes9xTYL/blob/mecm/Release/Releases.md

krazynez, 2022-03-28:
I have tried those versions and they work perfectly fine. I would need verification with someone that has a 2k/3k/GO to see if it is a non-1k issue.

McFly75, 2022-03-28:
Tested it on my PSP 2000 and same crash.

Was unable to test it on my PSP 1000 at the moment because firmware was to old, unable to run chronoswitch_7_0 because firmware to old, don´t know if it is safe to update using the official firmware 6.61 when it has CFW, but was unable to update it anyway, because it doesn't have a battery and when updating it says to wait for the battery to charge more before you can update.

McFly75, 2022-03-28:
And just for curiosity sake, I also tried it on my PSP Street (E-1000) and same crash.
So have basically tried every PSP except PSP 1000.

krazynez, 2022-03-28:
Are you on the stable release or a beta?

McFly75, 2022-03-28:
Tried both.
What version should I try?

krazynez, 2022-03-28:
Both should work, was just curious. I will try to get some others to help test on other PSP's and see if they are having the same issue. I will report back with any results that they find.

Demonic098, 2022-03-29:
@McFly75 just unlock extra memory from ARK-4 recovery menu (on models higher than 1K)
tested on 2K 3K ARK-4.12

McFly75, 2022-03-29:
YES, now it works. Thank You.
Enabled extra memory for Homebrew.

closed 2022-03-29 by McFly75

JoseAaronLopezGarcia, 2022-03-29:
Ah yes, if the homebrew detects that your model has extra RAM and tries to use it but the CFW hasn't enabled it then it will crash.
It is highly advised to always have the "High Mem" option set to "homebrew" for maximum compatibility and stability.

McFly75, 2022-03-29:
Yes, good advise.
Never had this problem on Pro-C2 CFW.
Maybe "High Mem" is enabled by default?
Only thing I can find in Recovery Menu is:
Advanced, Force High Memory Layout (Enabled or Disabled)
And heard that this may cause problems.

Only need a fix for VSH plugins (Category Lite) getting disabled when using "Pause Game" on a PSP Go.
Then I will switch permanently to Ark-4 cfw.