PSP-Archive/ARK-4 #18: PSP Go Pause game & Categoy_lite does not work.

McFly75, 2022-03-24:
When using Pause Game (Save Sate/Ram dump) & with Category_lite, games are missing and need a soft reboot in order to work.

McFly75, 2022-03-26:
Did some more testing, it seems that all VSH plugins gets disabled when using Pause Game.
But the VSH menu still works.
A soft reboot enables the VSH plugins again.

I like the Pause game function on the PSP Go.
It makes it possible to use a "save state" for PS1 games, which is not possible any other way.
And it makes it quick to resume a PS1 or a PSP Game (skipping intro etc.)

JoseAaronLopezGarcia, 2022-04-07:
Can you try with latest release?
https://github.com/PSP-Archive/ARK-4/releases/tag/r145

McFly75, 2022-04-07:
Same problem, VSH plugins gets disabled when using Pause Game function.
Ark-4.14.5

And got a new problem when running Ark_Live 4.14.5
I lost the VSH menu (Press Select) - I have opened a New Issue thread.

JoseAaronLopezGarcia, 2022-04-08:
The latest ARK automatically disables pause game when it deems it necessary (when you use highmem or inferno cache).
I recommend you leave the option disabled so it doesn't load on VSH (you don't need it there), and just let ARK automatically disable it on games since you will most likely already have inferno cache enabled for them (recommended).

McFly75, 2022-04-08:
Not sure what you mean?
Which option should I disable?
And what does inferno cache do?

The Option I have enabled in Recovery Menu is:
Unlock Extra Memory: Homebrew (in order for Snes9xTYL to work)
Memory Stick Speedup: Always (faster read/write speeds?)
Inferno Cache: Game (not sure what it does, but only PSP games uses Inferno driver or does PS1/homebrew use the driver?)

pmastrosimone, 2023-01-10:
So I had the same issue, but it is a ~~bug~~ feature inherent to all CFW's (at least both PRO and ARK), and @JoseAaronLopezGarcia 's response about the Inferno cache made me realize, and after testing I did confirm. I think because when pausing a game, the ISO is unloaded (as the game is "closed"), loading it back up will cause the system to crash (basically it will run through whatever is left in RAM/the stack, about 2 secs of gameplay in my testing) before it needs to load from the ISO, but since it's not loaded, it crashes. ARK generally blocks this (no crash, no pause), whereas PRO does not (crash after restore). To solve both you can convert your ISOs to PBPs, or just disable Pause Game. To my knowledge, Inferno doesn't cache PBPs, so games may be slower. For me, I'd rather have the pause game (and slightly more space) resultant from PBPs, but you may prefer the speed of ISOs.

McFly75, 2023-01-11:
On my PSP GO with "661_PRO-C2_14-02-2015" , I never had a problem with Pause game.
Both PSP in CSO & ISO format works fine & PBP format for PS1 also works fine.
So PRO-C2 does not crash after restore on my PSP GO.

JoseAaronLopezGarcia, 2023-01-11:
> On my PSP GO with "661_PRO-C2_14-02-2015" , I never had a problem with Pause game. Both PSP in CSO & ISO format works fine & PBP format for PS1 also works fine. So PRO-C2 does not crash after restore on my PSP GO.

Maybe you have inferno cache disabled?

McFly75, 2023-01-11:
The PRO-C2 CFW does not have Inferno cache (I dont think).
I believe that only Ark-4 has Inferno cache.

Should the Pause function work on Ark-4 with both PSP & PS1 games, if I disable the Inferno cache?

McFly75, 2023-01-11:
OK, I just found the Inferno cache on the PRO-C2 firmware.
Never knew it was there and never used it (has always been disabled)
What does it do anyway? Speed up loading of games?

It does indeed cash PSP games when enabled and loading a Paused game.
And it does work when loading a PS1 game in PBP format. (i don´t believe PS1 games uses Inferno at all)

JoseAaronLopezGarcia, 2023-01-12:
> OK, I just found the Inferno cache on the PRO-C2 firmware. Never knew it was there and never used it (has always been disabled) What does it do anyway? Speed up loading of games?
>
> It does indeed [...]

Yeah PS1 doesn't use Inferno. Only ISO/CSO, not even PBP.
The cache is supposed to work similar to the UMD cache by Sony. It's supposed to speed up loading in some games but at this point in time I'm sure it's mostly useless if you have a decent memory stick and a faster format like JSO or ZSO.

JoseAaronLopezGarcia, 2023-01-12:
> I also am now starting to have the plugin loading issue. I either didn't notice it or it did not happen before. Plugins loaded with the "vsh" switch are not reloaded after a "Pause Game" independent of game type or ARK settings. Plugins marked "always" still work as expected, as does the "Quit Game" plugin loading. Does not appear to be relegated to certain plugins (in my case both dayviewer and GC_Lite exhibit this behavior). If necessary I can upload my ARK settings to determine the cause of the issue

Interesting, the only thing that comes to mind is that when resuming from a game the VSH is loaded with a different apitype. Which would explain why it works with "always" (since that one doesn't check the apitype).
Right now the VSH switch checks for apitypes 0x210 and 0x220 but perhaps there's another one that needs to be checked.

pmastrosimone, 2023-01-12:
Not sure if [this](https://github.com/codestation/gclite) would help. Without being familiar with PSP plugins I cannot determine apitype, that being said I'll add some things about my config that *may* be pertinent (or completely useless), my plugins are on ef0:/, all of my games are on ms0:/, and I have plugin compatibility "on". Not sure if that redirection undergoes any odd behavior when plugins are loaded, I will try to mess around with moving a game to ef0:/ and/or messing around with plugin compatibility

JoseAaronLopezGarcia, 2023-01-12:
Is this error exclusive to ARK or does PRO also have it?

pmastrosimone, 2023-01-12:
In my limited testing it appears ARK-exclusive, both dayviewer and gc_lite are loaded after a pause operation, I will try messing around with settings (e.g., the ef0/ms0 distinction) later tonight and report back. If there are other things I should be messing with do let me know

JoseAaronLopezGarcia, 2023-01-12:
Just as I suspected, the issue was that when returning from the game after pausing, the apitype is different (0x300) from expected (0x210/0x220).
Issue should be fixed now.

JoseAaronLopezGarcia, 2023-01-12:
Here's new release fixing this issue:
https://github.com/PSP-Archive/ARK-4/releases/tag/r4195

closed 2023-01-12 by JoseAaronLopezGarcia

pmastrosimone, 2023-01-13:
Confirmed, thanks for your help!