PSP-Archive/ARK-4 #23: Pacman World 3 crashes the PSP (on Release 14.1, no plugins, works with Pro-C2)

McFly75, 2022-04-03:
Pacman World 3 crashes the PSP (on Release 14.1, no plugins, works with Pro-C2)
Tested on a PSP 3000 and a PSP GO, with plugins disabled, tried both cso and iso.
When starting the game, screen turns black and PSP shots down after a few seconds.
Works fine with PRO-C2 cfw, don´t know why???

krazynez, 2022-04-05:
DEBUG version PSPLINK output 4.14.4 
```
host0:/> syspatch: PMA(0x0123)
Loading all modules ... Ready
0: 
bkHostAppendFile: Open host0: failed - sleeping
bkHostAppendFile: Open host0: failed - sleeping
bkHostAppendFile: Open host0: failed - sleeping
bkHostAppendFile: Open host0: failed - sleeping
bkHostAppendFile: Open host0: failed - sleeping
bkHostAppendFile: Open host0: failed - sleeping
bkHostAppendFile: Open host0: failed - sleeping
bkHostAppendFile: Open host0: failed - sleeping
.....
```

krazynez, 2022-04-05:
update: Works from ArkMenu (Custom Launcher) looking into why it won't from xmb however.

krazynez, 2022-04-05:
latest pre-release fixes this issue: https://github.com/PSP-Archive/ARK-4/releases/tag/v1649190604

McFly75, 2022-04-07:
Weird, but will still not work on my PSP 3000 (both cso & iso)
v1649245439

JoseAaronLopezGarcia, 2022-04-07:
Can you try with latest release?
https://github.com/PSP-Archive/ARK-4/releases/tag/r145

McFly75, 2022-04-07:
Tried Version 4.14.5, and it still crashes the PSP 3000.
But it does work fine when launched using the AKR-4 Custom Launcher.

McFly75, 2022-04-07:
Sorry my bad, Pacman World 3 works fine now on Version 4.14.5
I was only replacing the SAVEDATA ARK_01234
& not replacing/running ARK_Live again.

closed 2022-04-07 by McFly75