PSP-Archive/ARK-4 #26: ARK 4.14.5 VSH Menu not showing up on PSP GO.

McFly75, 2022-04-07:
ARK 4.14.5 VSH Menu not showing up on PSP GO.
Ran Ark_Live/ARK-4.14.5 and not I have no VSH menu.
But ARK-4 is running.
- [X ] GO
- ARK 4.14.5

JoseAaronLopezGarcia, 2022-04-07:
I'm experiencing the same issue and similar issues with 1K. I'm looking into it.

krazynez, 2022-04-07:
very strange if I use a pre-release after 4.14.5 its working fine. But nothing has changed for vshmenu since then.

McFly75, 2022-04-07:
Replaced FLASH0.ARK from v1649356105 (as it seemed to be the only file changed from 4.14.5 to v1649356105)
and ran ARK_live and now VSH menu is back.

krazynez, 2022-04-07:
@McFly75 you read my mind, was just typing if you wanted to test a pre-release

JoseAaronLopezGarcia, 2022-04-08:
https://github.com/PSP-Archive/ARK-4/releases/tag/r146

McFly75, 2022-04-08:
FIXED (Version 4.14.6)

closed 2022-04-08 by McFly75