PSP-Archive/ARK-4 #27: Pacman World 3, Sleep crashes the PSP

McFly75, 2022-04-07:
Pacman World 3, when entering sleep mode, close PSP Go or quick flip power switch on PSP 3000.
Green light flashing a lot and then the PSP shuts down.
Tested with ARK-4.14.5 and a PSP 3000 and a PSP GO.
(Other games enter sleep mode fine, at least the ones I have tested so far)

3000 (3k/slim & brite)
GO
ARK-4.14.5

krazynez, 2022-04-07:
Anywhere in the game? Like a level or even the main menu? 

Just tried on a 1k and resumes fine for both, are you sleeping then instantly resuming or waiting a bit then resuming?

McFly75, 2022-04-07:
At the very first level, after "killing" the cardboard monsters and then killing the real mosters, I pause the game (not crossing the bridge or anything) I put the PSP to sleep and it starts flashing green several times and then turns off, so no change to wake up the PSP.
And it happen on both my PSP 3000 and PSP GO.

krazynez, 2022-04-07:
Okay, I just replicated the steps, yet no sign of crashing/shutting down... hmmm. Okay I will have to ask someone with a 3K & GO to test. Thanks for the info :+1:

McFly75, 2022-04-07:
OK.
I seems that it doesn't matter when you put the game to sleep, it will always crash the PSP.
(and I haven't found any other game that does this)

JoseAaronLopezGarcia, 2022-04-08:
It's weird, I can't get to reproduce the bug.
Using JSO format with 8K block size on PSP 1K with MS Speedup and Inferno Cache enabled.
Game plays fine, there's barely any IO on the memory stick, FMVs play smoothly, sleep mode works fine at any point.

McFly75, 2022-04-08:
I don't know why.
I have tried iso, cso, different image.
And when going to sleep, the PSP shuts down.
(I even tried to convert iso to jso with 8k block size and it still crashes)
Using 4.14.6
(This does not happen on Pro-C2 so I don't think there is anything wrong with the image file)

McFly75, 2022-04-08:
Unfortunately I'm unable to test this on my PSP 1K because I have no working battery for it and it has an old firmware installed and I can not update to 6.61 without a battery (PSP firmware checks for battery level when updating firmware)

I do have:
PSP 1K (No battery + Old firmware)
PSP 2K
PSP 3K
PSP Street
PSP GO
PS Vita

JoseAaronLopezGarcia, 2022-04-08:
Any plugins or settings enabled?

McFly75, 2022-04-08:
When I test, I usually rename PLUGINS.TXT so I am sure no plugins are enabled.
And setting:
Unlock Extra Memory: Homebrew
Memory Stick Speedup: Always
Inferno Cache: Game

But I have tried with all settings disabled.

McFly75, 2022-04-08:
I think I have found out what is making the game crash.

Brightness Plugin

The weird thing is that I have only had this problem with Pacman World 3, other games work fine with the Brightness plugin enabled (and it also works fine on PRO-C2 cfw)

McFly75, 2022-04-08:
"Fixed" by adding the following line to PLUGINS.TXT
ULUS10055, ms0:/seplugins/brightness.prx, off
(That way I can still use Brightness plugin for all other games)

JoseAaronLopezGarcia, 2022-04-09:
Oh yeah I've had reports of this happening on other games with this same plugin, which is the reason I implemented the ability to disable plugins on specific games.

closed 2022-04-09 by JoseAaronLopezGarcia