PSP-Archive/ARK-4 #31: [feature request] ArkLauncher: Filebrowser Startup Option

svensvond, 2022-06-13:
## ARK-4 Bugs/Issues

#### PSP Model
- [ ] 1000 (1k/Phat)
 
- [ x] 2000 (2k/Slim)

- [ ] 3000 (3k/slim & brite)

- [ ] GO

- [ ] Street (E-1000)  

#### Ark Version
4.17

Add an option to go directly to the filebrowser.

Why?
If you have a large microSD Card + many ISOs on the card, categorizing in folders is a great way to handle the iso amount. Typically the large sd card takes some time to init load - not with the ark-launcher. Here i can load the isos in almost no time, since meta data and the sd card "itself" is not loaded. Ark-Launcher works like a charm for that - would be nice to go directly to the file browser on startup of ark-launcher. Games are categorized like iso\action, iso\sports .. etc.

JoseAaronLopezGarcia, 2022-06-17:
This is a good idea, I will work it out as a setting for the custom launcher.

JoseAaronLopezGarcia, 2022-07-14:
This feature will be available in the next update.

closed 2022-07-14 by JoseAaronLopezGarcia

JoseAaronLopezGarcia, 2022-07-14:
https://github.com/PSP-Archive/ARK-4/releases/tag/v1657825892