PSP-Archive/ARK-4 #32: Support for games that can only use WEP/Open AP over infrastructure

MojoJojoDojo, 2022-06-14:
On normal Adrenaline its not possible and the game will hang when trying to get to the AP connection screen.

JoseAaronLopezGarcia, 2022-06-17:
How do I reproduce the issue in i.e. Twisted Metal Head On?

MojoJojoDojo, 2022-06-19:
@JoseAaronLopezGarcia you go to the online infrastructure mode ingame and the AP dialog wont show up.
In that point the game will hang.
Tbh I didnt try this for myself and had someone trying it with ARK CFW awhile ago.
Reported it didnt work for him

Zero1UP, 2022-07-11:
(Vita only issue) As stated games that can't do anything other than WEP / OPEN when going into infrastructure mode won't work.

In the case of ATV Offroad Blazin Trails it will just bounce you back to the Multiplayer menu where you select what mode you want to connect to. (AD HOC / Infra)

JoseAaronLopezGarcia, 2022-07-14:
Do you know if Adrenaline has a fix for this issue? Or is it something that hasn't been developed yet?
I will check it out and see what I can come up with, but if any of you have any information that you think will be useful please don't hasitate to share.

MojoJojoDojo, 2022-07-15:
Well it doesnt work on Adrenaline and I don't it will be fixed there because the development of it sorta got halted...
Thats why i opened an issue here to see if its possible to fix it via CFW.

MojoJojoDojo, 2022-07-15:
> but if any of you have any information that you think will be useful please don't hasitate to share.

The games that can only use WEP/Open seem to load the infrastructure AP menu by using a prx file from inside the game.
That's what I noticed when I used umdgen to figure it out.

JoseAaronLopezGarcia, 2023-01-13:
> > but if any of you have any information that you think will be useful please don't hasitate to share.
> 
> The games that can only use WEP/Open seem to load the infrastructure AP menu by using a prx file from inside the game. That's what I noticed when I used umdgen to figure it out. Maybe there is a way to bypass loading it from the game and instead load it from the firmware tor something... (Which will also help PSPs use WPA for the game). But because it loads that prx file for the menu maybe its just hardcoded to use only that.

So perhaps the fix would be to hook that prx loading and load your own that reimplements the features.
Sounds like something that can be done in a plugin so it could be compatible with ARK and Adrenalinea, rather than a built-in ARK feature, but it's also something that would take time to create.
Perhaps a new project can come out of this.

Pik-a-boo, 2023-01-13:
> Same applies to adding WPA2 to the PSP. It requires reversing the WiFi firmware and making a custom one. Something that is completely outside the scope of ARK and should be a separate project (therefore also compatible with all CFWs).

Yeah but here you got something here that doesn't work at all on a Vita while it works fine on a PSP (unless its the Street model).
It looks more like a "fix" then an "enhancement" here...
I know someone is already working on a plugin that could solve this though.

JoseAaronLopezGarcia, 2023-06-06:
Here's a little interesting detail that anybody can try out:
- Trying to connect to wifi on the XMB in PS Vita (i.e. Adrenaline/ARK) will make it behave like a PSP and only let you connect to Open, WEP or WPA but not WPA2.
- However if you try to connect using the Custom Launcher, it will connect to whatever your PS Vita is configured for (even a WPA2).

I'm guessing it's a matter of the utility parameters so maybe we can do something to make games that do the first behave like games that do the second.

Pik-a-boo, 2023-06-06:
Just a note but the games will also have issues even if they are chovy signed unlike with the other USB issue.

JoseAaronLopezGarcia, 2024-10-28:
> Thinking about how #409 was solved, by version spoofing to get the OSK to show the new QWERTY layout on old software, couldn't a similar thing be occurring here where software compiled with an SDK predating WPA support (like WipeOut Pure and the other examples listed before) has WPA hidden/blocked. EDIT: Nah, tried version spoofing on any relevant prx to networking and it did nothing, possibly because of these launch titles loading ancient PRX versions from within the UMD filesystem instead.

It's certainly a good theory, no idea how to go about this though, would have to figure out which module is doing such filtering and there's quite a few net modules.

NirYeF, 2024-11-04:
I know someone that managed to create a niteptr cheat to allow you to connect to all APs for the game ATV Offroad Fury: Blazin' Trails.
Not sure if it could have helped with the issue on the Vita but on PSP it could allow you to connect to WPA at least.
If someone wants the cheat its here : 
```
#Patch Netconf Init
; ATV US Linblow
; Allows connections to all APs
; Routine
0x004046a0 0x27bdfff0
0x004046a4 0xafbf0000
0x004046a8 0x34080044
0x004046ac 0xac880000
0x004046b0 0x34080000
0x004046b4 0xac880038
0x004046b8 0x34080000
0x004046bc 0x0e300e99
0x004046c0 0xac880040
0x004046c4 0x8fbf0000
0x004046c8 0x03e00008
0x004046cc 0x27bd0010
; Change alloc size for SceUtilityNetconfParam to 0x44 (ALIGN(0x44, 16) = 0x50)
0x00313d88 0x34040050
; Change netconf param size from 0x38 to 0x44
0x00313e38 0x34060044
; Hook
0x00313ef0 0x0e3011a8
```
game id is `UCUS-98603.txt`.

Double-0-seven7, 2025-06-24:
Should be fixed/workedaround by using the new atpro plugin here by Kethan : https://github.com/Kethen/aemu/releases/tag/2025-06-24
Might be a good idea to use the code and implement it in ark4 itself like you did with the wpa2 code because when you use the wpa2 option it actually crashes these games.
I only tried 1 game but it also works with another game I know.