PSP-Archive/ARK-4 #33: [Bug] Cannot launch infinity (2.05) from ArkLauncher

svensvond, 2022-06-14:
## ARK-4 Bugs/Issues

#### PSP Model
- [ ] 1000 (1k/Phat)
 
- [x] 2000 (2k/Slim)

- [ ] 3000 (3k/slim & brite)

- [ ] GO

- [ ] Street (E-1000)  

#### Ark Version

- 4.17

#### Steps to Reproduce
1. go to ark-launcher
2. run infinity in psp\game\update

Expected: Infinity (2.05) is loading
Behavior: Default PSP XMB is loaded

JoseAaronLopezGarcia, 2022-07-03:
Dupplicate of this:
https://github.com/PSP-Archive/ARK-4/issues/35

Also please note that infinity is different project, it would be ideal if you ask there for propper ARK support.
https://github.com/DaveeFTW/Infinity

closed 2022-07-03 by JoseAaronLopezGarcia

svensvond, 2022-07-07:
Actually the latest release of the infinity project is 2.03 (3 years ago, just one master/main branch) - you are shipping with 2.05. Who made the update?

JoseAaronLopezGarcia, 2022-07-08:
> Actually the latest release of the infinity project is 2.03 (3 years ago, just one master/main branch) - you are shipping with 2.05. Who made the update?

I made the update, with a very buggy SDK, which was generating really bad ARK builds and is probably responsible for all the Infinity crashes. I changed SDK to an older one since then, but now it's too old to compile Infinity. My guess is only Davee would be able to generate a proper build for Infinity.

JoseAaronLopezGarcia, 2022-08-29:
Note: infinity builds in newer ARK releases are no longer buggy.