PSP-Archive/ARK-4 #35: [Bug] İnfinity 2.0.5 crushes and psp shuts down on Ark cfw

exousia61, 2022-06-26:
## ARK-4 Bugs/Issues

#### PSP Model
- [ ] 1000 (1k/Phat)
 
- [ ] 2000 (2k/Slim)

- [X] 3000 (3k/slim & brite)

- [ ] GO

- [ ] Street (E-1000)  

#### PSP Firmware
- [X] 6.61

#### Ark Version

- 4.17.1

#### Steps to Reproduce
1. After installing fresh 6.61
2. go to ark launcher & wait for install
3. run infinity 2.0.5
4. then, psp shuts down

JoseAaronLopezGarcia, 2022-06-26:
First run and install infinity (from OFW), then run and install ARK, then run infinity again (in ARK) to configure autoboot.

exousia61, 2022-06-26:
I did what you said but the results are the same
first i installed infinity
then install ark from ark loader
and tried to run infinity, but psp shutted down again

stolzerrabe, 2022-07-02:
Same issue here on  PSP 2000 (2k/Slim)

I´d ran CFW 6.61 Pro-c Infinity before.

Tried to Install ARK on Top of Pro...
So i just followed the Installguide and Installed the ARK_0123 to Savegame, ARK_Live to Game and Infinity to UPDATE.
Installed ARK, Choose Ark as Module in Infinity. Restart -> PRO runs.
Tried to Open Infinity again -> Black PSP -> Shutdown
This i tried several Times.

Then i decided to run Chronoswitch to go back to OFW.
Installed Infinity, Installed ARK, Reopen Infinity for Autoboot... -> Black PSP -> Shutdown

All i can use is AKR without Infinity =(

stolzerrabe, 2022-07-03:
Ok, i got it run.
When i tried to start infinity with OFW (Hold Home on Start) it makes PSP Crash.
As i run it on CFW i was able to update Infinity.
Then i selected ARK as Module. Restart. Gone =)

I really believe that in my first tries yesterday, i forgot to update infinity and just ran it and switched the Module to Ark. So it always crashed^^

JoseAaronLopezGarcia, 2022-07-03:
Please note that Infinity is different project from ARK and with a different author. It would be best if you ask for support there.

closed 2022-07-03 by exousia61

leleobhz, 2022-08-03:
> Please note that Infinity is a different project from ARK and with a different author. It would be best if you ask for support there.

Hello @JoseAaronLopezGarcia 

As cited in [Infinity official site](https://infinity.lolhax.org/), Infinity isn't being mantained. Also, you applied a patch to allow ARK to be pinned on Infinity, so the specific patch from 2.0.3 to 2.0.5 maybe discussed here since its a fork.

That said, I incurred in this issue in my PSP 2000. I had success with follow procedure:

- [Chronoswitch 7.0](https://github.com/DaveeFTW/Chronoswitch/releases/tag/v7.0) to flash OFW 6.61
- After first boot, replace OFW EBOOT.PBP to patched 2.0.5 Infinity2
- Do the first Infinity Install
- Reboot and open ARK-4
- From ARK-4, launch [Chronoswitch 7.0](https://github.com/DaveeFTW/Chronoswitch/releases/tag/v7.0)  again to patch Infinity2 (Will show Frm 6.61 to 9.99)
- Reboot twice and check VSH menu.

Someone here can check if this procedure applies as general instruction? @JoseAaronLopezGarcia can you please reopen the issue?