PSP-Archive/ARK-4 #38: Issue with the game Driver 76

MojoJojoDojo, 2022-07-17:
## ARK-4 Bugs/Issues

#### PSP Model
- [ ] 1000 (1k/Phat)
 
- [ ] 2000 (2k/Slim)

- [ ] 3000 (3k/slim & brite)

- [X] GO

- [ ] Street (E-1000)  

#### Ark Version

- Enter Ark Release here (press Select it shows the version at the top of vshmenu). If using beta build put commit version.
- [7d39dcfa029783922dbcbe99b40a320281ef12e0](https://github.com/PSP-Archive/ARK-4/releases/tag/v1657850931)

#### Homebrew/GAME (if applicable)

- If using retail game please add GAME ID i.e: UCUS-98713 
Driver 76 ULUS10235
#### Plugins ( if applicable )

 - Please list all plugins enabled here.
adhocredirector.prx / CWUSB (https://github.com/codedwrench/CWUSB/) (not enabled at the same time)
npdrm_free plugin (if using PBP).
#### Steps to Reproduce
1.enable either CWUSB plugin or adhocredirecotor plugin
2.go into the game
3.go into multiplayer
4.the game will freeze for 3-4 secs while loading it when you use an ISO/CSO , the game will get out with the error 80010087 into VSH (With PBP)

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

Other notes:
Old plugins on PSP Go is enabled in settings.

MojoJojoDojo, 2022-07-17:
Note that I didn't encounter this sort of thing on PRO CFW when I tried the game before.

JoseAaronLopezGarcia, 2022-07-17:
Can you try with the latest stable release (4.17.1) here: https://github.com/PSP-Archive/ARK-4/releases/tag/r171
I need to get an idea if this bug has been there forever or if it's due to the new inferno driver.
I remember playing this game some time back and not having issues, but I didn't test multiplayer either.

Also, try enabling Inferno cache, some games require it to function properly (most notably Twisted Metal Head On, but might apply to others as well). NOTE: on PRO, Inferno Cache is always enabled by default, so it's a good idea to enable it in ARK too so both CFW are on equal grounds, ARK simply doesn't enable it by default, which woud give the impression that it works worse than PRO.

MojoJojoDojo, 2022-07-17:
> Can you try with the latest stable release (4.17.1) here: https://github.com/PSP-Archive/ARK-4/releases/tag/r171 I need to get an idea if this bug has been there forever or if it's due to the new [...]

I had inferno cache disabled on pro though.
But I will try what you said later

MojoJojoDojo, 2022-07-17:
Tried with inferno cache on but it still crashes...
Weird thing is is that you can get in with an ISO/CSO but the plugins will just stop working at that point.
Can i downgrade to stable without using chronoswitch? currently i have it with Infinite installed.

MojoJojoDojo, 2022-07-18:
Hmm can't run infinite as it gives me a error so I guess i will have to use chronoswitch again sigh.
Not gonna do it for now then.

JoseAaronLopezGarcia, 2022-07-19:
There's no need to use chronoswitch.
Replace ARK_01234 savedata folder with older release, turn on your PSP holding the Home button and launch ARK Loader to install ARK again from the older release.

JoseAaronLopezGarcia, 2022-08-07:
To fix this issue I will have to make a patch to disable np9660 cache. I'll let you know when it's available.