PSP-Archive/ARK-4 #40: Dayviewer V7 Plugin Does Not Work Properly

TheSubPlayer, 2022-08-02:
## ARK-4 Bugs/Issues

The PSP crashes and shut down while the plugin is active

#### PSP Model
- [ ] 1000 (1k/Phat)
 
- [ ] 2000 (2k/Slim)

- [X] 3000 (3k/slim & brite)

- [ ] GO

- [ ] Street (E-1000)  

#### Ark Version

- ARK 4.18

#### Plugins 

vsh, ms0:/seplugins/dayviewer.prx, off
vsh, ms0:/seplugins/Npdrmfree/npdrm_free.prx, on
vsh, ms0:/seplugins/category_lite.prx, on
game, ms0:/seplugins/Cushome/cushome.prx, on
game, ms0:/seplugins/CwCheat/cwcheat.prx, on
pops, ms0:/seplugins/popsloader/popsloader.prx, on
pops, ms0:/seplugins/popsloader/cdda_enabler.prx, on
all, ms0:/seplugins/Prxshot/prxshot.prx, on
all, ms0:/seplugins/RemoteJoyLite/RemoteJoyLite.prx, on
ULUS10160, ms0:/seplugins/cheatdevice_remastered.prx, on
ULUS10041, ms0:/seplugins/cheatdevice_remastered.prx, on
ULUS10280, ms0:/seplugins/mh2censo.prx, on

#### Steps to Reproduce

1. Install dayviewer.prx and dayviewer_config.txt in ms0:/seplugins
2. In PLUGINS.TXT, put "vsh, ms0:/seplugins/dayviewer.prx, on"

#### Recovery settings

- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [ ] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [X] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [ ] Disabled
	- [ ] Always
	- [X] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

[dayviewerV7.zip](https://github.com/PSP-Archive/ARK-4/files/9241440/dayviewerV7.zip)

JoseAaronLopezGarcia, 2022-08-07:
I will check it out. Thanks for the report.

One consideration about your plugins file:

You load cwcheat for all games (and homebrew too? I'd say better use "umd" instead of "game" so it only applies to actual games and not homebrew).

game, ms0:/seplugins/CwCheat/cwcheat.prx, on

But then you also load the gta cheat device on GTA games:
ULUS10160, ms0:/seplugins/cheatdevice_remastered.prx, on
ULUS10041, ms0:/seplugins/cheatdevice_remastered.prx, on

I recommend you disable cwcheat on gta games so it doesn't conflict with cheat device, by adding this line at the end of the plugins file (or anywhere after the line that enables cwcheat):
ULUS10160, ms0:/seplugins/CwCheat/cwcheat.prx, off
ULUS10041, ms0:/seplugins/CwCheat/cwcheat.prx, off

TheSubPlayer, 2022-08-07:
Yes, I've noticed this before, I was still learning how to use the new plugin system. 
Now I've learned how to better separate plugins. Thanks for the tip!

JoseAaronLopezGarcia, 2022-08-07:
Fixed:
https://github.com/PSP-Archive/ARK-4/releases/tag/r181

closed 2022-08-07 by JoseAaronLopezGarcia