PSP-Archive/ARK-4 #41: [Feature Request] Recovery Menu Additions

TheSubPlayer, 2022-08-02:
#### PSP Model
- [ ] 1000 (1k/Phat)
 
- [ ] 2000 (2k/Slim)

- [X] 3000 (3k/slim & brite)

- [ ] GO

- [ ] Street (E-1000)  

#### Ark Version

ARK 4.18

Request: Add FTP Server option in Recovery menu and disable Sony startup animation / Gameboot.

The FTP server in the recovery menu would be a good addition to File Manager, which is also available in recovery.

And the option to disable Sony startup and Gameboot Logo is self-explanatory ¯\_(ツ)_/¯.

JoseAaronLopezGarcia, 2022-08-07:
At this point I'm wondering about merging custom launcher and recovery and just let users have their own custom RECOVERY.PBP or RECOVERY.PRX if they want to.

xgwmir, 2022-08-13:
The poploader plug-in on psvita does not work properly

JoseAaronLopezGarcia, 2022-08-13:
> The poploader plug-in on psvita does not work properly

Well, first of all, PS1 games don't work on Vita with ARK, use Adrenaline for PS1 games, it works very well there.
Secondly, popsloader doesn't work on Vita, at all, not even in Adrenaline.
The PS1 emulator for PSP (pops) needs to be heavily patched to work on Vita, no popsloader plugin has this ability, and even if they do, the games would have no sound due to missing Media Engine coprocessor.

Getting PS1 games, or popsloader, to work with ARK on PS Vita is not a priority since it would never be as good as the official PS1 emulator for Vita (which is actually pops for PSP but improved and fixed by Sony to work on PS Vita).

JoseAaronLopezGarcia, 2022-08-22:
The feature to disable Sony startup and gameboot logo has been already added.
There are no plans to add FTP to recovery menu to keep this menu as simple as possible.

closed 2022-08-22 by JoseAaronLopezGarcia