PSP-Archive/ARK-4 #44: Items do not load automatically in "Game"if Files is the default menu.

svensvond, 2022-08-22:
## ARK-4 Bugs/Issues

If file-browser is selected as default menu, click on "Game" in the nav bar results in an empty section with no items.
You need to press "select" first to get items loaded.

#### Ark Version

- 4.18.7

JoseAaronLopezGarcia, 2022-08-23:
This is the intended behaviour, the idea of replacing the games menu with file browser is because you don't want the games menu to load automatically.
Also if games were to load automatically, it'll make the file browser very slow since the PSP is busy processing all the IO and games list.

svensvond, 2022-08-23:
Make sense. Maybe i have a special workflow than.
Would it be possible to load the apps/games when switching to games browser?
I usually put my games in a different subfolder, so that they dont intefere with the games browser. Iam using the games browser to launch apps / differen cfw though.

JoseAaronLopezGarcia, 2022-08-23:
Alright that's a good solution, I've updated the latest release with this fix:
https://github.com/PSP-Archive/ARK-4/releases/tag/r1810

closed 2022-08-23 by JoseAaronLopezGarcia

svensvond, 2022-08-24:
Thanks a lot!