PSP-Archive/ARK-4 #46: [Feature Request] Option to force Inferno driver for PSN (EBOOT.PBP) games

Jake1702, 2022-08-25:
This is something I've been interested in for the longest while, essentially the inverse of what the Galaxy controller did, and as I finally switched to ARK today I figured I might as well inquire about it. Perhaps utilizing Inferno over Sony's NP9660 driver could have some small performance benefits?

Jake1702, 2022-08-25:
I'll admit I was mostly interested due to the fact that there are a few games that are notably smaller as EBOOTs than CSOs (and DAX as well). The three that immediately spring to mind are the 2 Midnight Club games and Coded Arms, each saving a couple hundred MBs give or take. (I may be the only person left on the planet who considers that a very notable space saving, and I have an unhealthy obsession with storage space efficiency. Coincidentally, it's a similar situation with PS1 EBOOTs, but due to I assume different reasons.)

However, I was under the impression that the format was already mostly understood seeing as we can decrypt them and acquire the contained ISO as well as being able to package ISOs into signed EBOOTs that run on OFW. Perhaps I was mistaken?

JoseAaronLopezGarcia, 2022-08-26:
I understand but if these games work fine with Sony's driver, what would be the benefit of making a homebrew one?
The amount of work to do it would be way too big and the benefits would be zero, specially considering that it would require lots of optimizations to run as fluent as the original.

I really don't see the need for this feature other than wasting time and overbloating ARK with experimental features that might not be very noob-friendly. Sony's driver works perfectly fine for PBP format games and Inferno works perfectly fine for homebrew formats, and users don't have to worry about having to select a driver.

JoseAaronLopezGarcia, 2022-08-26:
I will be closing this since I doubt it will ever go anywhere. The idea is interesting but very time consuming and wouldn't really bring anything new or better.

closed 2022-08-26 by JoseAaronLopezGarcia