PSP-Archive/ARK-4 #48: [Feature Request / Bug] Ark Menu exit to VSH, Games exit to Ark Menu

svensvond, 2022-08-26:
## ARK-4 Bugs/Issues
Possibility to exit to VSH from within arkmenu - but exit games to arkmenu.
The recovery option "Autoboot Launcher" does not work as expected.
Always / VSH exit to launcher - all other options to vsh - from game and arkmenu.

#### Ark Version
4.x.x.x

JoseAaronLopezGarcia, 2022-08-26:
I don't understand how is this a bug, this is the expected behaviour.
Enabling autoboot launcher (set to always) completely disables VSH and replaces it with the custom launcher.

svensvond, 2022-08-26:
Ok, if thats intended, than this is just a feature request.
It would be nice to quit the launcher although it is set to autoboot. 
The exit button on the arklauncher launches the launcher again - maybe you could implemented an easy way to exit to vsh?

JoseAaronLopezGarcia, 2022-08-26:
Yeah I understand but it's not that easy. When you exit using the custom launcher button it is actually exiting to VSH, but when VSH is about to start it gets redirected to custom launcher because the settings file has set autoboot to always. The only way to prevent this is to disable autoboot.
The whole idea of autoboot is to fully replace VSH, if you want to have VSH and custom launcher at the same time then simply disable the autoboot. If you're in a game and want to exit to the custom launcher instead of the XMB you can use the key combo L+R+Start+Down.

svensvond, 2022-08-26:
Thanks for the explanation - I get the problem.
Let me explain how there could be a work around - if you are willing to implement.
I usually exit to VSH by executing a different CFW (PRO/ME) directly. If we could make the exit button configureable to launch what ever app we want, we could "exit" to that app. I know it does not make sense if you do not have another CFW installed - but the exit button itself does not make sense if you set arklauncher to autorun.
So my idea is to add a custom button here - or use the exit button.
You might ask: Why?
I have several tasks that could by done only on the VSH - launching Infinity as a minor example is one of them (black screen if run from arklauncher), or go to the system settings for example.

JoseAaronLopezGarcia, 2022-08-26:
What do you mean you exit to VSH by executing another CFW? Why go through all the hassle?
If you use the VSH then why enable autoboot launcher? The feature is to literally replace the VSH, just disable it.

The exit button still makes sense in the autoboot mode as a way to refresh the custom launcher, it acts as a soft reboot. Lets say you install a new theme by replacing THEME.ARK in the savedata folder using the file browser, then you can use exit to restart the custom launcher so it reads the new theme.

I really don't understand why you are using autoboot launcher when you clearly don't need that feature, it seems like you are just limiting yourself for no reason. The autoboot feature is for people who don't like the XMB/VSH and just want their lists of games to show up and ready to boot.

svensvond, 2022-08-26:
I dont want you drive you mad - lets just think a second about my "workflow" - it its not useful for anyone, we could just close this issue - it simply stays a request.
I need the launcher set to autoboot - because its working great and much better than the VSH ever could. Filebrowsing and launching with fast loading times is a great feature to work with larger game collections.
Things i cant do within arklauncher:

- Go to system settings (iam currently doing hardware mods and i need to do that)
- Launch Infinity - to temporarily (!) activate another firmware, not using that often, but ark inifinity is buggy and cant be run from ark firmware

Actually thats it. Again: Priority low and i can live without that feature.

JoseAaronLopezGarcia, 2022-08-26:
Well the issue with launching infinity from the custom launcher is because Infinity is not a game/app but an updater but the custom launcher doesn't check that. This can be fixed.

As for going to the system setting, you can always go to recovery, disable autoboot, do whatever on VSH, go back to recovery and re-enable autoboot.

As for changing the behaviour of the autoboot itself, the real problem here is I don't want to risk breaking a feature just for a very specific use case that has easy workarounds (like going to recovery to temporarily disable the autoboot).

In any case I appreciate your requests and support and I'm glad you like the CFW.

What if we settle on a middleground? I will fix Infinity on the Custom Launcher so you can switch CFW easily from there.

svensvond, 2022-08-26:
Thanks - that would be even MUCH better than a configureable button or touching the autolaunch section!

JoseAaronLopezGarcia, 2022-08-26:
Alright I got it working, here you go:
https://github.com/PSP-Archive/ARK-4/releases/tag/r1813

closed 2022-08-26 by JoseAaronLopezGarcia