PSP-Archive/ARK-4 #51: Infinity does not load - PSP 3000 only

svensvond, 2022-09-02:
## ARK-4 Bugs/Issues
Infinity does not load from ark launcher or ark vsh on PSP 3000 models only.

#### PSP Model
- [ x] PSP-3000 (Slim & Bright)

#### Ark Version
4.18.13 - 18

#### Expected
Infinity is loading

#### Behaviour
Black screen, System turns off

#### Notes
Refers to Issue #48 - was fixed for (at least) PSP 2000 there.
Same Software and ark configuration works on a or multiple (tested 2) PSP 2000. 
Tested on 2 different PSP 3000s.

svensvond, 2022-09-05:
Same SD works in a PSP 2000, does Not on a 3000. Same ark Version installed on Same ofw.

JoseAaronLopezGarcia, 2022-09-05:
> Same SD works in a PSP 2000, does Not on a 3000. Same ark Version installed on Same ofw.

That's strange, I've used the same SD on different models with different OFW (6.60 or 6.61) and never had any issues, except maybe some plugins.
What exactly doesn't work?

svensvond, 2022-09-05:
Here you go: [PSP 2000 vs 3000](https://www.veed.io/view/d26f2d3b-f07d-432e-ab39-bbcd445a97e8)

JoseAaronLopezGarcia, 2022-09-06:
> Here you go: [PSP 2000 vs 3000](https://www.veed.io/view/d26f2d3b-f07d-432e-ab39-bbcd445a97e8) If you dont have a 3000, maybe you could build a debug version with logging?

I have a 3K and don't have any issues with Infinity. Why are you hot swapping the memory cards? That could be an issue.

svensvond, 2022-09-06:
Weird - i indeed have it running on a 3000 perfectly - and the other one does not. Same Firm and Software - and no: Not any "hot swapping" :) [3000vs3000](https://streamable.com/bnz323)

JoseAaronLopezGarcia, 2022-09-19:
> Weird - i indeed have it running on a 3000 perfectly - and the other one does not. Same Firm and Software - and no: Not any "hot swapping" :) [3000vs3000](https://streamable.com/bnz323) What i also did: Reset back to OFW with Chronoswitch, installed Infinity + Ark from the latest Pre-Release - no difference, infinity does not load from ark.

Run pspident on both PSP's to see if they are different motherboards.

svensvond, 2022-09-19:
There are some differences.
**Working System**
Mainboard: TA-090V2
Mainboard Type: Bright 3004 (03g)
PSP Version: v1.0
Possible CFW: 5.03 HEN and up

**NOT working System**
Mainboard: TA-095V2
Mainboard Type: Bright 3004 (09g)
PSP Version: v4.1
Possible CFW: 6.3x HEN and up

JoseAaronLopezGarcia, 2022-09-20:
Is the same true for older versions of Infinity?
Does ARK itself have any difference in behavior between those two models?

svensvond, 2022-09-21:
Ark seems to work fine, BUT
"Official" Infinity 2.03 (without the ARK autoboot option) does indeed work!

svensvond, 2022-09-21:
4.16 (just to be sure) also does not work.

ventod, 2022-10-05:
I have the same error on my PSP 3000 MHB.
PSP Everest 2 rev 5 reports TA-093v2, PSPident 0.77.777 reports TA-093 with no version.

PSPIdent 0.77.777
[image]

PSP Everest 2 rev 5
[image]

I've also tried to "clean" my psp first by downgrading to 6.30 and upgrading to 6.61 again.

It has to be said that I did change some settings before installing the CFW e trying to install infinity. I changed language (Japanese to English) and char set (to multi-language latin 1) and other minor stuffs, but I do not think this should make any difference.

ventod, 2022-12-26:
Small update:

Tested this in 4.19.1 and 4.19.2 and this is still an issue.

svensvond, 2022-12-26:
I can confirm that.

JoseAaronLopezGarcia, 2022-12-31:
So this issue only shows on a very specific 3K?

svensvond, 2022-12-31:
i dont think so - its not working on a range of 3000 systems, its more like the other way around.

Working System (x2 System tested)
Mainboard: TA-090V2
Mainboard Type: Bright 3004 (03g)
PSP Version: v1.0
Possible CFW: 5.03 HEN and up

NOT working System (x3 System tested)
Mainboard: TA-093
Mainboard Type: Bright 3004 (04g)
PSP Version: v3.0
Possible CFW: 6.2x HEN and up

NOT working System
Mainboard: TA-095V2
Mainboard Type: Bright 3004 (09g)
PSP Version: v4.1
Possible CFW: 6.3x HEN and up

krazynez, 2023-01-05:
> I'm having this same issue with my PSPGo, Infinity just crashes if launched from ARK, I did this from a clean firmware install via Chrono, then launched ARK and when Infinity is starting the console shuts down.
> 
> Also a lot of games wont launch with ARK -1
> 
> PSPIdent [image]

Can you open another Issue and be more specific on which games are not launching with ARK please.

twokilohertz, 2023-01-07:
I am experiencing the same issue as @ventod & @svensvond. I have an 04g PSP 3003 and the version of Infinity included with ARK-4 does not work. Black screen & system powers off.

JoseAaronLopezGarcia, 2023-01-09:
> **Working System** (x2 System tested) Mainboard: TA-090V2 Mainboard Type: Bright 3004 (03g) PSP Version: v1.0 Possible CFW: 5.03 HEN and up
> 
> **NOT working System** (x3 System tested) Mainboard: TA-093 Mainboard Type: Bright 3004 (04g) PSP Version: v3.0 Possible CFW: 6.2x HEN and up
> 
> **NOT working System** Mainboard: TA-095V2 Mainboard Type: Bright 3004 (09g) PSP Version: v4.1 Possible CFW: 6.3x HEN and up
> 
> Seems like TA-90V2 is the only one working.

So it affects models above 03g, that's interesting.
What's the behaviour when trying to launch ARK from OFW?

ventod, 2023-01-09:
4.19.3 Fixed this for me!
I do not know what changed. I installed ARK as always:

- Connected PSP to PC with USB
- Downloaded the last version of ARK
- From the archive copied the **ARK_Live** folder to **PSP/GAME/** (removed the older folder already present)
- From the archive copied the **ARK_01234** folder to **PSP/SAVEDATA/** (removed the older folder)
- From the archive copied the **Infinity/EBOOT.PBP** file to **PSP/GAME/UPDATE/** (removed the older folder)
- Safely removed USB device from windows
- Run the ARK installer > boot to ARK 4.19.3 Live > checked this in both system settings and VSH menu
- Connect to PC and removed the **FLASH0.ARK** file from **/PSP/SAVEDATA/** (as the guide on the main page suggests)
- Safely removed USB device from windows
- Shutdown the PSP
- Boot the PSP (OFW)
- Run the Infinity installer
- It didn't crash!
- Install infinity
- Set to boot ARK automatically
- Done!

JoseAaronLopezGarcia, 2023-01-09:
> 4.19.3 Fixed this for me! I do not know what changed. I installed ARK as always:
>
> * Connected PSP to PC with USB
> * Downloaded the last version of ARK
> * From the archive copied the **ARK_Live** [...]

This is very strange, I guess we'll have to wait for others to post their results.

FrncoDz, 2023-01-09:
What [ventod](https://github.com/ventod) mentioned also worked for me, I can run Infinity from OFW without any issues and turn Infinity to 6.61∞, but running it directly on ARK crashes as normal. Launching ARKLive on 6.61∞ also crashes the Go. I'll try with newest ARK version and report.

JoseAaronLopezGarcia, 2023-01-09:
> What [ventod](https://github.com/ventod) mentioned also worked for me, I can run Infinity from OFW without any issues and turn Infinity to 6.61∞, but running it directly on ARK crashes as normal. Launching ARKLive on 6.61∞ also crashes the Go. I'll try with newest ARK version and report.

Running ARK Live on OFW with infinity crashes? Now that's even worse.
I do remember having to install Infinity the first time from OFW and the second time, to configure it, from ARK.

JoseAaronLopezGarcia, 2023-01-09:
I'm not entirely sure how this bug happens, here's what I usually do to install ARK-4 and Infinity:

- Power on PSP with OFW.
- Install Infinity.
- Install ARK-4.
- Run infinity again to configure ARK-4 autoboot.

Maybe this procedure doesn't work for everyone. There's probably some bug in the Infinity for ARK-4 build but I can't compile it myself any more.

svensvond, 2023-01-09:
@JoseAaronLopezGarcia 
I think running Infinity from OFW or another CFW work - cant confirm it right now. 
You cant compile inifinity anymore?

JoseAaronLopezGarcia, 2023-01-09:
> @JoseAaronLopezGarcia I also think running Infinity from OFW or another CFW works - cant confirm it right now. You cant compile inifinity anymore?

No I can't, the SDK I use (for compiling ARK-4) is very old (it's the same SDK used to compile PRO and ME way back in the day). It isn't compatible with Infinity's C++ language standards.
And even if I could compile, I'm not sure I would be able to properly debug and fix the issue, best would be to try to get in contact with the original developer.

twokilohertz, 2023-01-09:
>     * Power on PSP with OFW.
> 
>     * Install Infinity.
> 
>     * Install ARK-4.
> 
>     * Run infinity again to configure ARK-4 autoboot.

This is the method I ended up using as if one boots ARK-4 first, then tries to launch Infinity 2.0.5 - that will fail and crash the system. However installing Infinity first and _then_ ARK-4 works seemingly fine. 

I gave building the project myself a go and as of tag r4192 it works completely fine. When I've got some time I might give relieving us from the python2 dependency and the old PSPSDK dependency a go.

JoseAaronLopezGarcia, 2023-01-09:
> This is the method I ended up using as if one boots ARK-4 first, then tries to launch Infinity 2.0.5 - that will fail and crash the system. However installing Infinity first and _then_ ARK-4 works [...]

That would be nice but I tried using a newer SDK and the results were binary files that are almost double the ones that the older SDK generated and thus resulting in memory issues and bugs all over the place.
I did not go any further into figuring out if I was using outdated makefile rules and flags or whatnot, so maybe it's possible.

JoseAaronLopezGarcia, 2023-01-11:
I've updated the readme to better point out that Infinity has to be installed from OFW, and then configured from ARK.
If this solves the issues OP reported we can close it.

svensvond, 2023-01-11:
Does ark somehow run the file differently from the other (older) CFWs or OFW? Some kind of different call maybe?

JoseAaronLopezGarcia, 2023-01-11:
> Does ark somehow run the file differently from the other (older) CFWs or OFW? Some kind of different call maybe?

It shouldn't, the app is launched as an updater by the XMB. Maybe there's a broken patch in the XMB somewhere.
It would be interesting to see if the same happens when launching Infinity from the Custom Launcher.

JoseAaronLopezGarcia, 2023-01-13:
If the workaround has solved the issue I'm gonna close this. Feel free to reopen it if it's still a thing.

closed 2023-01-13 by JoseAaronLopezGarcia