PSP-Archive/ARK-4 #52: CWCheat doesnt work - gameid is wrongly detected

a0a0max, 2022-09-02:
## ARK-4 Bugs/Issues

#### PSP Model

- [ ] PSP-E1000 (Street)
Not tested on other systems

#### Ark Version

72760699c159320e64260e7b58e77da3fb7ca213

#### Homebrew/GAME (if applicable)
Tested several psp games (e.g. Tekken, Burnout, etc.)

#### Plugins ( if applicable )
CWCheat

#### Steps to Reproduce
1. Start a game
2. Keep Select pressed to show CWCheat
3. CWCheat is shown
4. Game id detected by CWCheat seems wrong (probably associated to CWCheat itself instead of psp game)
5. No cheat can be selected

All recovery options are disabled

a0a0max, 2022-09-11:
It was cwcheat 2.3, the latest version. Unfortunately I have no opportunity to perform tests in these weeks

JoseAaronLopezGarcia, 2022-12-31:
Has this been solved?

closed 2023-01-09 by JoseAaronLopezGarcia