PSP-Archive/ARK-4 #57: Comments are deleted in PLUGINS.TXT after enabling/disabling plugins via recovery

TheSubPlayer, 2022-12-13:
I know you're aware of the error, and I understand that you're quite busy.
I'm just registering it here because it's a very useful resource for organizing PLUGINS.TXT, and it's nothing really important or urgent ¯\_(ツ)_/¯ 

See ya soon

krazynez, 2022-12-14:
[This](https://github.com/PSP-Archive/ARK-4/blob/9d96c58ee4e6c5cdc563684a918ea9b4334cddf5/core/systemctrl/src/plugin.c#L215) should be working for plugins. Which type are you trying? `//`, `#` or `;`. Can you test using another type of comment and see if you get the same results or not.

TheSubPlayer, 2022-12-15:
I've tested with all of them, same result

JoseAaronLopezGarcia, 2023-01-09:
Fixed in Version 4.19.3
https://github.com/PSP-Archive/ARK-4/releases/tag/r4193

closed 2023-01-09 by JoseAaronLopezGarcia