PSP-Archive/ARK-4 #59: Read the installation instructions again. You have to copy the K.BIN file from VitaBubble folder into ARK_01234 folder.

Windows10User2, 2022-12-18:
Part 1: https://github.com/PSP-Archive/ARK-4/issues/55#issuecomment-1356506823

That's precisely what I did. Also, I tried deleting the ARK_01234 folder created by FastARK and also only replaced its files for the downloaded ARK_01234 folder and I'm having the same issues in both scenarios.

Windows10User2, 2023-01-14:
> Nobody else is encountering this problem. Perhaps you should make a video showing the steps you are taking to install ARK so that we can see where the issue might be.

Actually, there is someone having this issue like one can read [here](https://gbatemp.net/threads/release-ark-4-cfw-port-for-psp.581263/page-40#post-10028749) and [here](https://gbatemp.net/threads/release-ark-4-cfw-port-for-psp.581263/page-41#post-10037832).

JoseAaronLopezGarcia, 2023-01-16:
> > Nobody else is encountering this problem. Perhaps you should make a video showing the steps you are taking to install ARK so that we can see where the issue might be.
> 
> Actually, there are some users also having this issue like one can read [here](https://gbatemp.net/threads/release-ark-4-cfw-port-for-psp.581263/page-40#post-10028749) and [here](https://gbatemp.net/threads/release-ark-4-cfw-port-for-psp.581263/page-41#post-10037832).

both of those users eventually got it right by following instructions, you're still the only one I know unable to get anything working at all. I can't do anything unless you make a detailed description of what you're doing so I can see the issue.

Windows10User2, 2023-01-16:
> both of those users eventually got it right by following instructions, you're still the only one I know unable to get anything working at all. I can't do anything unless you make a detailed [...]

Since when did they got it right? Anyway, like I told before, after running a PSP game (I only have ISOs in the PSP, PS Vita and PS TV), I have the C1-2858-3 error and ARK closes (this is when ARK works on the PS TV). On the PS TV, after running ARK, it crashes often with that error (C1-2858-3) during the initial screen with a black background and white letters stating kernel exploit failed so one is unable to even reach the part where one chooses the game he/she wants to play (on the PS Vita at least one reaches it). Also, speaking about instructions, if it wasn't me, one wouldn't even know he/she supposedly had to install a PSN game before installing ARK.

JoseAaronLopezGarcia, 2023-01-17:
> Since when did they got it right? Anyway, like I told before, after running a PSP game (I only have ISOs in the PSP, PS Vita and PS TV), I have the C1-2858-3 error and ARK closes (this is when ARK [...]

Most of those people had issues back when firmware 3.65+ wasn't supported, or they didn't correcty installed K.BIN, they all got it eventually.
"Kernel exploit failed" means you didn't install it correctly, follow the instructions and copy the correct K.BIN file for PS Vita.

Windows10User2, 2023-01-17:
> > > > > Nobody else is encountering this problem. Perhaps you should make a video showing the steps you are taking to install ARK so that we can see where the issue might be.
> > > > 
> > > > 
> > > > Actually, there are some users also having this issue like one can read [here](https://gbatemp.net/threads/release-ark-4-cfw-port-for-psp.581263/page-40#post-10028749) and [here](https://gbatemp.net/threads/release-ark-4-cfw-port-for-psp.581263/page-41#post-10037832).
> > > 
> > > 
> > > both of those users eventually got it right by following instructions, you're still the only one I know unable to get anything working at all. I can't do anything unless you make a detailed description of what you're doing so I can see the issue.
> > 
> > 
> > Since when did they got it right? Anyway, like I told before, after running a PSP game (I only have ISOs in the PSP, PS Vita and PS TV), I have the C1-2858-3 error and ARK closes (this is when ARK works on the PS TV). On the PS TV, after running ARK, it crashes often with that error (C1-2858-3) during the initial screen with a black background and white letters stating kernel exploit failed so one is unable to even reach the part where one chooses the game he/she wants to play (on the PS Vita at least one reaches it). Also, speaking about instructions, if it wasn't me, one wouldn't even know he/she supposedly had to install a PSN game before installing ARK.
> 
> Most of those people had issues back when firmware 3.65+ wasn't supported, or they didn't correcty installed K.BIN, they all got it eventually. "Kernel exploit failed" means you didn't install K.BIN correctly, follow the instructions and copy the correct K.BIN file for PS Vita.

I already did and how do you know their issues were fixed if they didn't say it? Also, I always installed them in the 3.60 FW.

JoseAaronLopezGarcia, 2023-01-17:
> I already did and how do you know their issues were fixed if they didn't say it? Also, I always installed them in the 3.60 FW.

The error is very clear, you haven't properly installed K.BIN from VitaBubble folder into ARK_01234 folder.
It's as simple as that.

JoseAaronLopezGarcia, 2023-01-17:
What do you mean "what about the C1-2858-3 error"?
I've said it many times before and this is the last one: if you are unable to follow simple instructions, then it's not an issue I (or anyone) can resolve.

I'm closing this and any further duplicate of this since nobody can confirm this as an actual issue.

closed 2023-01-17 by JoseAaronLopezGarcia