PSP-Archive/ARK-4 #61: PSP 1003 - 4.19, CIPL - Custom Launcher Recovery Mode Option Freezes

Dark98, 2022-12-19:
Trying to launch the Recovery menu through the Custom Launcher results in the attached screenshot, works fine when running through the XMB VSH menu

JoseAaronLopezGarcia, 2022-12-20:
Is this still an issue in latest version (4.19.1)?

Dark98, 2022-12-21:
Yeah, still effects 4.19.1

I've only tested a cIPL Environment if that makes any difference

Demonic098, 2022-12-23:
I cannot reproduce that behavior
For me it's forking just fine
PSP1K TA-081 FW 661 ARK-4.19.1 cIPL.
@Dark98 Can you send more details plz.

JoseAaronLopezGarcia, 2022-12-25:
I can't seem to reproduce the issue either. Are there any particular steps?

Dark98, 2022-12-26:
I used chronoswitch to reflash a stock 6.60 firmware then ran ark & used the cIPL flasher..
No necessary steps, just happens when trying to launch recovery through the Custom launcher

Demonic098, 2023-01-22:
@Dark98 have you tried latest release ?

Dark98, 2023-01-22:
> @Dark98 have you tried latest release ?

same issue, just tested.. it worked on a previous version(i cant remember which)

JoseAaronLopezGarcia, 2023-02-12:
Is this still a thing? I can't recall if we've been able to reproduce it.

krazynez, 2023-02-17:
I can not reproduce this either. I just flashed 6.60 again on my Phat and then ran ARK 4.19.11 then cIPL Flasher, went into Custom Launcher and ran Recovery with no issues.

Dark98, 2023-02-18:
@JoseAaronLopezGarcia @krazynez Unsure what changed but I just tested with the latest build(Hadn't tested since my last comment) & It's working Normally! :)

closed 2023-02-18 by Dark98