PSP-Archive/ARK-4 #66: Github CI failure

krazynez, 2023-01-10:
Latest commit 97680e9dd8d3f5c30316dbd6e7939a582a57a620 causing CI build failure

`virtual_pbp.c:957: warning: ‘get_ISO_shortname’ defined but not used`

krazynez, 2023-01-10:
` psp-gcc -I/usr/local/pspdev/psp/sdk/include/libc -I/home/runner/work/ARK-4/ARK-4/common/include -I/home/runner/work/ARK-4/ARK-4/core/systemctrl/include -I. -I/usr/local/pspdev/psp/sdk/include -std=c99 -Os -G0 -Wall -fno-pic -D_PSP_FW_VERSION=660   -c -o hibernation_delete.o hibernation_delete.c
make[1]: *** No rule to make target 'imports.o', needed by 'vshctrl.elf'.  Stop.`

Latest commit: 5807177b9b505154b054fb8b9ae95ee69d01bb27

JoseAaronLopezGarcia, 2023-01-11:
There was a missing file, should be fixed now.

closed 2023-01-11 by JoseAaronLopezGarcia