PSP-Archive/ARK-4 #68: Unable to run v

Windows10User2, 2023-01-14:
When running Spider-Man: Web of Shadows – The Amazing Allies Edition (ULES01174), I have an error (80010016).

JoseAaronLopezGarcia, 2023-01-16:
Can you be a bit more detailed in your reports?
What format are you using? ISO, CSO, PBP, etc
Is it a decrypted PBP or encrypted PBP?

Windows10User2, 2023-01-16:
> Can you be a bit more detailed in your reports? What format are you using? ISO, CSO, PBP, etc Is it a decrypted PBP or encrypted PBP?

ISO. What do you mean by decrypted or encrypted PBP?

JoseAaronLopezGarcia, 2023-01-17:
What are you doing to break so many things?
I just tried the game and had no issues at all.

TheSubPlayer, 2023-01-17:
I also tested the game and it's working normally, do you have any plugins active?

JoseAaronLopezGarcia, 2023-01-17:
Closing this since it's not condfirmed to be ARK's issue.

closed 2023-01-17 by JoseAaronLopezGarcia

Windows10User2, 2023-01-17:
> What are you doing to break so many things? I just tried the game and had no issues at all.

Nothing, it's the only game that doesn't work and other users complaing about the other problems as well.

> I also tested the game and it's working normally, do you have any plugins active?

No.

> Closing this since it's confirmed to not be ARK's issue.

Since when?

JoseAaronLopezGarcia, 2023-01-18:
>
> Since when?
> 

Since I and others have tested it and found it to be no issue at all.
The issue is on your end, perhaps a corrupted ISO or memory card.

Windows10User2, 2023-01-18:
> > Since when?
> 
> Since I and others have tested it and found it to be no issue at all. The issue is on your end, perhaps a corrupted ISO or memory card.

I mean, since when did they state their issues were fixed?

I tried a lof of different ISOs and microSD cards/SD2VITAs.

JoseAaronLopezGarcia, 2023-01-18:
> > > Since when?
> > 
> > 
> > Since I and others have tested it and found it to be no issue at all. The issue is on your end, perhaps a corrupted ISO or memory card.
> 
> I mean, since when did they state their issues were fixed?
> 
> I tried a lof of different ISOs and microSD cards/SD2VITAs.

SD2Vita? Didn't you report the issue for PSP3000?

They stated the issues (with 3.65) were solved when I updated ARK to support 3.65 a while back.
The issue you are getting is because you are not installing K.BIN file correctly and at this point I'm not willing to continue this conversation as it's not related to any issues within ARK itself.

Windows10User2, 2023-01-18:
> SD2Vita? Didn't you report the issue for PSP3000?
>
> They stated the issues (with 3.65) were solved when I updated ARK to support 3.65 a while back. The issue you are getting is because you are not [...]

Sorry, I didn't mean SD2VITAs but microSD adapters. It's for the PSP 3000. How can I fix this issue?

What +3.65 issues on the PS Vita/PS TV are you talking about since I never had that FW version (I always had the 3.60 one) installed and I still have these issues (the boot one and the C1-2858-3 ones) since the beginning?

How can I not be installing it the right way since I deleted the ARK_01234 folder created by FastARK, transferred the downloaded one to the PS Vita and replaced its K.BIN file for the one in the downloaded VitaBubble folder and now it worked for the PS TV and not for the PS Vita? It is related to ARK unlike what you say.

Windows10User2, 2023-01-24:
After replacing the  "–" in the ISO filename with a "-", I stopped having the issue which prevented me from playing this game [Spider-Man: Web of Shadows – The Amazing Allies Edition (ULES01174)].