PSP-Archive/ARK-4 #70: Emulators won't work when using ARK

lorenzofelletti, 2023-01-19:
## ARK-4 Bugs/Issues
I cannot run games using the emulators: the emulator starts, but when I press on a ROM the PSP freezes and shuts down.
However, if I use PRO Fast Recovery the games runs fine.
Am I doing something wrong? I think I did the installation correctly, so I don't know where the problem lies...

#### PSP Model
- [ ] PSP-1000 (Fat)
 
- [ ] PSP-2000 (Slim)

- [X] PSP-3000 (Slim & Bright)

- [ ] PSP-N1000 (PSP Go)

- [ ] PSP-E1000 (Street)

- [ ] PCH-1000/1100 (Vita Fat)

- [ ] PCH-2000 (Vita Slim)

- [ ] VTE-1000 (Vita TV)

#### Ark Version

- 4.19.6

#### Homebrew/GAME (if applicable)

- all emulators (gPSP, snes9xTYL, ...) 

#### Steps to Reproduce
1. open an emulator
2. try to run a game

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

JoseAaronLopezGarcia, 2023-01-19:
Which emulator is it? (Can you provide a link?)
Also have you tried enabling Extra RAM in recovery menu?

lorenzofelletti, 2023-01-19:
Thank you for the rapid answer. Extra memory solved the issue!

Btw, the emulators were gpSP and snes9xTYL, unfortunately I don't have the links anymore...

JoseAaronLopezGarcia, 2023-01-22:
Reopening because this is a bug. ARK should be able to detect homebrew that use extra RAM and should automatically enable it.

JoseAaronLopezGarcia, 2023-01-22:
This has been fixed and it is no longer necessary to enable High Memory for these homebrews to work.
https://github.com/PSP-Archive/ARK-4/releases/tag/r4197

closed 2023-01-22 by JoseAaronLopezGarcia