PSP-Archive/ARK-4 #71: ARK crashes at first PS Vita restart after installing it but works in the following times

GitHubCFW, 2023-01-22:
After installing ARK, it crashes after restarting the PS Vita and running ARK saying it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error, but after closing it and running ARK again and/or restarting the PS Vita the following times and running it again, ARK doesn't crash, so apparently I only have this issue once. It's odd. I'm on 3.60.

[image]

[image]

JoseAaronLopezGarcia, 2023-01-23:
The kernel exploit we're using right now is a bit unstable in nature. It works most of the time, but it might fail depending on how RAM is. Which is why a fresh restart makes it work again. It should be possible to make it more stable, so I'll look into it.

GitHubCFW, 2023-01-23:
> > After installing ARK, it crashes after restarting the PS Vita and running ARK saying it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error, but after closing it and running ARK again and/or restarting the PS Vita the following times and running ARK again, it doesn't crash, so apparently I only have this issue once. It's odd. I'm on 3.60.
> > [image]
> > [image]
> > My PSP 3000 turns off if having the Inferno Cache, Extra Memory and Memory Stick speedup settings enabled when running a PSP game. I disabled the Inferno Cache and kept the Extra Memory and Memory Stick speedup settings enabled and I was able to run PSP games without the PSP turning off.
> 
> Finally, someone with the same problem as me but according to the developer, it's my fault.
> 
> > The kernel exploit we're using right now is a bit unstable in nature. It works most of the time, but it might fail depending on how RAM is. Which is why a fresh restart makes it work again. It should be possible to make it more stable, so I'll look into it.
> 
> Didn't you say it was my fault for not getting it to work?

I don't know about you but regarding the first issue, I always followed the instructions and I had this problem on the PS TV from the start and recently I've been having it on the PS Vita too.

Regarding the second one, I didn't have it in the previous version.

krazynez, 2023-01-23:
> Finally, someone with the same problem as me but according to the developer, it's my fault.
>
>
> Didn't you say it was my fault for not getting it to work?

> Perhaps you should make a video showing the steps you are taking to install ARK so that we can see where the issue might be. (closed issue #59 )

He asked you to provide visual representation of the issue you were having. ARK-4 is vast and wide we do not always know what generic error codes are, just better to know when and where they're happening.

JoseAaronLopezGarcia, 2023-01-23:
> > > After installing ARK, it crashes after restarting the PS Vita and running ARK saying it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error, but after closing it and running ARK again and/or restarting the PS Vita the following times and running ARK again, it doesn't crash, so apparently I only have this issue once. It's odd. I'm on 3.60.
> > > [image]
> > > [image]
> > > My PSP 3000 turns off if having the Inferno Cache, Extra Memory and Memory Stick speedup settings enabled when running a PSP game. I disabled the Inferno Cache and kept the Extra Memory and Memory Stick speedup settings enabled and I was able to run PSP games without the PSP turning off.
> > 
> > 
> > Finally, someone with the same problem as me but according to the developer, it's my fault.
> > > The kernel exploit we're using right now is a bit unstable in nature. It works most of the time, but it might fail depending on how RAM is. Which is why a fresh restart makes it work again. It should be possible to make it more stable, so I'll look into it.
> > 
> > 
> > Didn't you say it was my fault for not getting it to work?
> 
> This is NOT at all the same issue you reported and certainly not in the same way, your reports were generic, badly written, lacked any content and were nothing more than mere rants. You also somehow reported the same issue back when we were using a different kernel exploit for 3.60 so its impossible that you are reporting the same thing as OP. Refrain from saying anything useless from now on.

Windows10User2, 2023-01-23:
> > > After installing ARK, it crashes after restarting the PS Vita and running ARK saying it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error, but after closing it and running ARK again and/or restarting the PS Vita the following times and running ARK again, it doesn't crash, so apparently I only have this issue once. It's odd. I'm on 3.60.
> > > [image]
> > > [image]
> > > My PSP 3000 turns off if having the Inferno Cache, Extra Memory and Memory Stick speedup settings enabled when running a PSP game. I disabled the Inferno Cache and kept the Extra Memory and Memory Stick speedup settings enabled and I was able to run PSP games without the PSP turning off.
> > 
> > 
> > Finally, someone with the same problem as me but according to the developer, it's my fault.
> > > The kernel exploit we're using right now is a bit unstable in nature. It works most of the time, but it might fail depending on how RAM is. Which is why a fresh restart makes it work again. It should be possible to make it more stable, so I'll look into it.
> > 
> > 
> > Didn't you say it was my fault for not getting it to work?
> 
> > Perhaps you should make a video showing the steps you are taking to install ARK so that we can see where the issue might be. (closed issue #59 )
> 
> He asked you @Windows10User2 to provide visual representation of the issue you were having. ARK-4 is vast and wide we do not always know what generic error codes are, just better to know when and where they're happening.

But GitHubCFW didn't provide a video too and he/she wasn't reprimended unlike me. The users I mentioned didn't provide one too.

> After installing ARK, it crashes after restarting the PS Vita and running ARK saying it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error, but [...]

How come it's not the same issue if I mentioned a boot issue and the error he's/she's having isthe same as mine too? Are you talking about your answers or the instructions you wrote? First you state I'm not having his/her issue and now you state I am? Again, how can my issue not be the same as his/hers if I described it exactly the way the pictures he/she uploaded (I stated it happened during boot when all those letters were showing up on the screen) and the error he's/she's having is the same as mine too? Regarding what you state about 3.60 and his/her issue not being the same as mine, it looks like you didn't read carefully but he/she said he/she bas been having it on the PS TV from the start so how can you be sure my issue is different from his/hers?

GitHubCFW, 2023-01-23:
> > > > After installing ARK, it crashes after restarting the PS Vita and running ARK saying it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error, but after closing it and running ARK again and/or restarting the PS Vita the following times and running ARK again, it doesn't crash, so apparently I only have this issue once. It's odd. I'm on 3.60.
> > > > [image]
> > > > [image]
> > > > My PSP 3000 turns off if having the Inferno Cache, Extra Memory and Memory Stick speedup settings enabled when running a PSP game. I disabled the Inferno Cache and kept the Extra Memory and Memory Stick speedup settings enabled and I was able to run PSP games without the PSP turning off.
> > > 
> > > 
> > > Finally, someone with the same problem as me but according to the developer, it's my fault.
> > > > The kernel exploit we're using right now is a bit unstable in nature. It works most of the time, but it might fail depending on how RAM is. Which is why a fresh restart makes it work again. It should be possible to make it more stable, so I'll look into it.
> > > 
> > > 
> > > Didn't you say it was my fault for not getting it to work?
> > 
> > 
> > > Perhaps you should make a video showing the steps you are taking to install ARK so that we can see where the issue might be. (closed issue #59 )
> > 
> > 
> > He asked you @Windows10User2 to provide visual representation of the issue you were having. ARK-4 is vast and wide we do not always know what generic error codes are, just better to know when and where they're happening.
> 
> But GitHubCFW didn't provide a video too and he/she wasn't reprimended unlike me. The users I mentioned didn't provide one too. But how can it be a generic error code if I stated the error I was having?
> 
> > > > > After installing ARK, it crashes after restarting the PS Vita and running ARK saying it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error, but after closing it and running ARK again and/or restarting the PS Vita the following times and running ARK again, it doesn't crash, so apparently I only have this issue once. It's odd. I'm on 3.60.
> > > > > [image]
> > > > > [image]
> > > > > My PSP 3000 turns off if having the Inferno Cache, Extra Memory and Memory Stick speedup settings enabled when running a PSP game. I disabled the Inferno Cache and kept the Extra Memory and Memory Stick speedup settings enabled and I was able to run PSP games without the PSP turning off.
> > > > 
> > > > 
> > > > Finally, someone with the same problem as me but according to the developer, it's my fault.
> > > > > The kernel exploit we're using right now is a bit unstable in nature. It works most of the time, but it might fail depending on how RAM is. Which is why a fresh restart makes it work again. It should be possible to make it more stable, so I'll look into it.
> > > > 
> > > > 
> > > > Didn't you say it was my fault for not getting it to work?
> > > 
> > > 
> > > This is NOT at all the same issue you reported and certainly not in the same way, your reports were generic, badly written, lacked any content and were nothing more than mere rants. You also somehow reported the same issue back when we were using a different kernel exploit for 3.60 so its impossible that you are reporting the same thing as OP. Refrain from saying anything useless from now on.
> 
> How come it's not the same issue if I mentioned a boot issue and the error he's/she's having is the same as mine too? Are you talking about your answers or the instructions you wrote? First you state I'm not having his/her issue and now you state I am? Again, how can my issue not be the same as his/hers if I described it exactly the way the pictures he/she uploaded (I stated it happened during boot when all those letters were showing up on the screen) and the error he's/she's having is the same as mine too? Regarding what you state about 3.60 and his/her issue not being the same as mine, it looks like you didn't read carefully but he/she said he/she bas been having it on the PS TV from the start so how can you be sure my issue is different from his/hers?

But at least I provided photos of the issue, unlike you. Anyway, apparently you're right when you say you have the same issue as I do because I've been having it since a previous version and I was always at 3.60 so unlike what #JoseAaronLopezGarcia says it is possible to have this issue for quite a long time and not just recently since I've had it for quite a while but I simply didn't report it.

Windows10User2, 2023-01-23:
> But at least I provided photos of the issue, unlike you. Anyway, apparently you're right when you say you have the same issue as I do because I've been having it since a previous version and I was [...]

But I thought I was clear enough since I mentioned a boot issue and the same error you are having. You even get we are having the same issue unlike @JoseAaronLopezGarcia so how could I have not been clear enough? Anyway, what matters is that we are having the same problem and for a while, not recently, unlike @JoseAaronLopezGarcia states.

JaegerFox, 2023-01-23:
let's see, did you make sure you have placed all the ARK1234 files in the savedata folder and replaced the k.bin file with the one in the vitabuble folder?

Windows10User2, 2023-01-23:
> let's see, did you make sure you have placed all the ARK1234 files in the savedata folder and replaced the k.bin file with the one in the vitabuble folder?

Are you talking to me? Yes, I did that, like always.

JaegerFox, 2023-01-23:
@Windows10User2 did I mention you? I wrote in the main issue in response to the initial question of @GitHubCFW

GitHubCFW, 2023-01-23:
> > > > > > After installing ARK, it crashes after restarting the PS Vita and running ARK saying it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error, but after closing it and running ARK again and/or restarting the PS Vita the following times and running ARK again, it doesn't crash, so apparently I only have this issue once. It's odd. I'm on 3.60.
> > > > > > [image]
> > > > > > [image]
> > > > > > My PSP 3000 turns off if having the Inferno Cache, Extra Memory and Memory Stick speedup settings enabled when running a PSP game. I disabled the Inferno Cache and kept the Extra Memory and Memory Stick speedup settings enabled and I was able to run PSP games without the PSP turning off.
> > > > > 
> > > > > 
> > > > > Finally, someone with the same problem as me but according to the developer, it's my fault.
> > > > > > The kernel exploit we're using right now is a bit unstable in nature. It works most of the time, but it might fail depending on how RAM is. Which is why a fresh restart makes it work again. It should be possible to make it more stable, so I'll look into it.
> > > > > 
> > > > > 
> > > > > Didn't you say it was my fault for not getting it to work?
> > > > 
> > > > 
> > > > > Perhaps you should make a video showing the steps you are taking to install ARK so that we can see where the issue might be. (closed issue #59 )
> > > > 
> > > > 
> > > > He asked you @Windows10User2 to provide visual representation of the issue you were having. ARK-4 is vast and wide we do not always know what generic error codes are, just better to know when and where they're happening.
> > > 
> > > 
> > > But @GitHubCFW didn't provide a video too and he/she wasn't reprimended unlike me. The users I mentioned didn't provide one too. But how can it be a generic error code if I stated the error I was having?
> > > > > > > After installing ARK, it crashes after restarting the PS Vita and running ARK saying it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error, but after closing it and running ARK again and/or restarting the PS Vita the following times and running ARK again, it doesn't crash, so apparently I only have this issue once. It's odd. I'm on 3.60.
> > > > > > > [image]
> > > > > > > [image]
> > > > > > > My PSP 3000 turns off if having the Inferno Cache, Extra Memory and Memory Stick speedup settings enabled when running a PSP game. I disabled the Inferno Cache and kept the Extra Memory and Memory Stick speedup settings enabled and I was able to run PSP games without the PSP turning off.
> > > > > > 
> > > > > > 
> > > > > > Finally, someone with the same problem as me but according to the developer, it's my fault.
> > > > > > > The kernel exploit we're using right now is a bit unstable in nature. It works most of the time, but it might fail depending on how RAM is. Which is why a fresh restart makes it work again. It should be possible to make it more stable, so I'll look into it.
> > > > > > 
> > > > > > 
> > > > > > Didn't you say it was my fault for not getting it to work?
> > > > > 
> > > > > 
> > > > > This is NOT at all the same issue you reported and certainly not in the same way, your reports were generic, badly written, lacked any content and were nothing more than mere rants. You also somehow reported the same issue back when we were using a different kernel exploit for 3.60 so its impossible that you are reporting the same thing as OP. Refrain from saying anything useless from now on.
> > > 
> > > 
> > > How come it's not the same issue if I mentioned a boot issue and the error he's/she's having is the same as mine too? Are you talking about your answers or the instructions you wrote? First you state I'm not having his/her issue and now you state I am? Again, how can my issue not be the same as his/hers if I described it exactly the way the pictures he/she uploaded (I stated it happened during boot when all those letters were showing up on the screen) and the error he's/she's having is the same as mine too? Regarding what you state about 3.60 and his/her issue not being the same as mine, it looks like you didn't read carefully but he/she said he/she bas been having it on the PS TV from the start so how can you be sure my issue is different from his/hers?
> > 
> > 
> > But at least I provided photos of the issue, unlike you. Anyway, apparently you're right when you say you have the same issue as I do because I've been having it since a previous version and I was always at 3.60 so unlike what @JoseAaronLopezGarcia says it is possible to have this issue for quite a long time and not just recently since I've had it for quite a while but I simply didn't report it.
> 
> But I thought I was clear enough since I mentioned a boot issue and the same error you are having. You even get we are having the same issue unlike @JoseAaronLopezGarcia so how could I have not been clear enough? Anyway, what matters is that we are having the same problem and for a while, not recently, unlike @JoseAaronLopezGarcia states.
> 
> By the way, do you have [this](https://github.com/PSP-Archive/ARK-4/issues/68) problem too?

Yes, I also have that problem. I just tested the game.

> let's see, did you make sure you have placed all the ARK1234 files in the savedata folder and replaced the k.bin file with the one in the vitabuble folder?

Yes, I did.

Windows10User2, 2023-01-24:
> @Windows10User2 did I mention you? I wrote in the main issue in response to the initial question of @GitHubCFW

Since when did you mention him/her?

krazynez, 2023-01-24:
Okay this isn't a chat room. Please stop with the bickering and lets move on and try to resolve the issue at hand. Thank you.

Windows10User2, 2023-01-24:
> Okay this isn't a chat room. Please stop with the bickering and lets move on and try to resolve the issue at hand. Thank you.

It wasn't me who started it, but @JoseAaronLopezGarcia.

JoseAaronLopezGarcia, 2023-01-25:
@GitHubCFW can you try this build to see if it fixes your issues with PSP 3000?
https://github.com/PSP-Archive/ARK-4/releases/tag/v1674650711

I just tried it myself and it works fine enabling all three settings (Inferno Cache, MS Speedup and Unlock Extra RAM).

As for the Vita's Kernel exploit stability, I am still looking into it, but it seems to break mostly when I've been playing other games (i.e. Vita games), probably small changes in the RAM layout.

GitHubCFW, 2023-01-25:
> @GitHubCFW can you try this build to see if it fixes your issues with PSP 3000? https://github.com/PSP-Archive/ARK-4/releases/tag/v1674650711
>
> I just tried it myself and it works fine enabling all [...]

@JoseAaronLopezGarcia, with that build I don't have this issue.

Regarding the other issue, I can't tell what you say since I only play PSP games. In my case it only happens once after the first restart.

Windows10User2, 2023-02-09:
> > @GitHubCFW can you try this build to see if it fixes your issues with PSP 3000? https://github.com/PSP-Archive/ARK-4/releases/tag/v1674650711
> > I just tried it myself and it works fine enabling all three settings (Inferno Cache, MS Speedup and Unlock Extra RAM).
> > As for the Vita's Kernel exploit stability, I am still looking into it, but it seems to break mostly when I've been playing other games (i.e. Vita games), probably small changes in the RAM layout.
> 
> @JoseAaronLopezGarcia, with the latest release, the PSP doesn't freeze and turn off immediatelly after running PSP games but it does after exiting them via the PS button. What is the difference between the DEBUG and the normal versions?
> 
> Regarding the boot issue, I can't tell what you say since I only play PSP games.
> 
> Also, I have a black screen on the PS Vita and PS TV if the Extra Memory setting is enabled.
> 
> EDIT: Now I'm having the boot issue often and not just after the first restart after installing and running ARK.

I'm also having the boot issue often and not just after the first PS Vita/PS TV restart after installing and running ARK.

GitHubCFW, 2023-02-12:
> > > @GitHubCFW can you try this build to see if it fixes your issues with PSP 3000? https://github.com/PSP-Archive/ARK-4/releases/tag/v1674650711
> > > I just tried it myself and it works fine enabling all three settings (Inferno Cache, MS Speedup and Unlock Extra RAM).
> > > As for the Vita's Kernel exploit stability, I am still looking into it, but it seems to break mostly when I've been playing other games (i.e. Vita games), probably small changes in the RAM layout.
> > 
> > 
> > @JoseAaronLopezGarcia, with the latest release, the PSP doesn't freeze and turn off immediatelly after running PSP games but it does after exiting them via the PS button. What is the difference between the DEBUG and the normal versions?
> > Regarding the boot issue, I can't tell what you say since I only play PSP games.
> > Also, I have a black screen on the PS Vita and PS TV if the Extra Memory setting is enabled.
> > EDIT: Now I'm having the boot issue often and not just after the first restart after installing and running ARK.
> 
> I'm also having the boot issue often and not just after the first PS Vita/PS TV restart after installing and running ARK.
> 
> By the way, you stated you had the boot issue since the first ARK-4 version. Which release are you talking about? @JoseAaronLopezGarcia states in GBAtemp it's impossible to have this issue on the previous ARK versions and on the latest ones.
> 
> Also, my PSP 3000 freezes and turns off if I have the Force Extra Memory and Inferno Cache settings set to Always when exiting PSP games using the PS button. Do you have a PSP 3000 to test? I'm asking you this because @JoseAaronLopezGarcia states in GBAtemp I'm the only one having this issue.

I'm talking about the January 2021 one, I think.

I have a PSP 3000 and tested it for you and in fact after setting to "Always" the Force Extra Memory and Inferno Cache settings and clicking "Yes" to exit PSP games using the PS button, the PSP freezes and shuts down.

JoseAaronLopezGarcia, 2023-02-12:
> I'm talking about the January 2021 one, I think.
>
> I have a PSP 3000 and tested it for you and in fact after setting to "Always" the Force Extra Memory and Inferno Cache settings and clicking "Yes" [...]

So let me get this straight, the procedure to recreate this issue is:
- Enable Inferno Cache and Force Extra Memory.
- Play a Game
- Try to exit.

@Windows10User2 was it so hard to explain this? You kept saying that another user had the same issue as you but the other user was talking about cwcheat. This kind of confusion is what makes it seem like you are lying even when you're not. Next time please take your time to properly document and explain the steps to reproduce your issues.

JoseAaronLopezGarcia, 2023-02-12:
I can't seem to be able to reproduce this issue on a PSP Street with latest ARK.
Is there anything else I might be missing? Could it apply only to certain games?
We need more information about this issue. I'll have other testers take a look at it too.

Windows10User2, 2023-02-12:
> I'm talking about the January 2021 one, I think.
>
> I have a PSP 3000 and tested it for you and in fact after setting to "Always" the Force Extra Memory and Inferno Cache settings and clicking "Yes" [...]

So, unlike what @JoseAaronLopezGarcia stated, it is possible to have the boot issue since the first ever ARK-4 released.

> So let me get this straight, the procedure to recreate this issue is:
>
> * Enable Inferno Cache and Force Extra Memory.
> * Play a Game
> * Try to exit.
>
> @Windows10User2 was it so hard to explain this? [...]

Are you kidding me? That's precisely what I've been stating! It's not my fault if you don't get what I state when I stated the exact same steps as @GitHubCFW because he/she got what I stated right away unlike you! And I was talking about this issue and you started stating I report I also had this issue in cwcheat when I didn't even have (and I still don't have) it! I could say the same to you since you're confusing issues and didn't even get the simple steps to recreate this issue! 

Also, what do you have to say about the boot and C1-2858-3 errors @GitHubCFW states he's/she's had since the first release in January 2021 and that I have since a long time ago? I thought the new kernel was different!

> I can't seem to be able to reproduce this issue on a PSP Street with latest ARK. Is there anything else I might be missing? Could it apply only to certain games? We need more information about this issue. I'll have other testers take a look at it too.

I have it with every PSP fame I tried.

JaegerFox, 2023-02-12:
@Windows10User2  do you have a picture/video to prove all that? i mean those "issues"

Windows10User2, 2023-02-12:
> @Windows10User2 do you have a picture/video to prove all that? i mean those "issues"

No, because they're obvious. Moreover, I was the first to report a previous PSP freezing and shutting down issue that occurred right after running PSP games on GBAtemp, other users also reported it and the issue was fixed. No photos were posted by me or the other GBAtemp users. And most of the issues couldn't be proved by posting photos but by uploading videos, don't you think? I guess the only exceptions would be the boot and the C1-2858-3 error issues and even then posting videos instead of photos would be better.

Also, @GitHubCFW posted photos about the boot issue 3 weeks ago and has the issue been fixed it? It looks like they were very "usefull" like y'all have been stating.

JoseAaronLopezGarcia, 2023-02-12:
> > @Windows10User2 do you have a picture/video to prove all that? i mean those "issues"
> 
> No, because they're obvious.

It's not that obvious when other people can't reproduce the issue. I can't reproduce it, my testers can't reproduce it.
You haven't mentioned anything at all to help reproduce it.

Does it happen with ISO, CSO or PBP?
PBP is known to not work well with force extra ram (Sony's official ISO driver makes use of the extra ram and thus they interfere). But of course you haven't said anything about the format so your report is USELESS.
If you're gonna keep doing this I will just close this issue and any other issue you open.

Windows10User2, 2023-02-12:
> It's not that obvious when other people can't reproduce the issue. I can't reproduce it, my testers can't reproduce it. You haven't mentioned anything at all to help reproduce it.
>
> Does it happen [...]

Then why can @GitHubCFW reproduce it right away, then? What about you, he/she reported the boot issue 3 weeks ago and even posted photos about it and what have you done to fix it?

Like I stated on GBAtemp, I only have ISOs. It's not my fault you weren't paying attention to what I reported. Moreover, @GitHubCFW didn't state the format he/she was having trouble with on the PSP and you didn't reprimend him/her like me! This issue wasn't even opened by me and you state that because you don't have arguments to what I state!

JaegerFox, 2023-02-12:
@Windows10User2 ok... I'll just mention that I sent you a video where I installed ARK-4 from scratch, and tested it by playing a few games... no problems. and You? only words 
Didn't you mention that you were no longer going to use ARK-4? why are you coming back?

JaegerFox, 2023-02-12:
@Windows10User2 you always have to go by what others say, you don't even know how to provide information on your own.

GitHubCFW, 2023-02-12:
> So let me get this straight, the procedure to recreate this issue is:
>
> * Enable Inferno Cache and Force Extra Memory.
> * Play a Game
> * Try to exit.
>
> @Windows10User2 was it so hard to explain this? [...]

To be honest, I have to agree with @Windows10User2 here. This PSP issue is very obvious and easy to recreate and I understood right away what he meant. I don't think it needed photos unlike the boot issue which is harder to describe. I recreated the issue right away after he mentioned the steps.

JoseAaronLopezGarcia, 2023-02-12:
> Then why can @GitHubCFW reproduce it right away, then? What about you, he/she reported the boot issue 3 weeks ago and even posted photos about it and what have you done to fix it?

Dude I work on this for FREE on my spare time, you are in no position to demand anything, and just for saying this I don't want you writing anymore on any of ARK's threads. Go find help somewhere else with that attitude.

closed 2023-02-12 by JoseAaronLopezGarcia

Windows10User2, 2023-02-12:
> @Windows10User2 ok... I'll just mention that I sent you a video where I installed ARK-4 from scratch, and tested it by playing a few games... no problems. and You? only words Didn't you mention that you were no longer going to use ARK-4? why are you coming back?