PSP-Archive/ARK-4 #74: [Feature Request] Per-Game settings

pmastrosimone, 2023-02-09:
This may already be a feature (though if it does its not in the docs), but being able to specify per-title/game settings in Settings.txt (or some other file) (especially overclock and high memory) would be nice

JoseAaronLopezGarcia, 2023-02-09:
As you correctly deduced, it is already a feature not mentioned in the docs.
The settings file uses the same format and same parser as the plugins file.
You can indeed do something like this:

ULUS01234, overclock, on

Doesn't work for disabling though (I might check this to see how it can be implemented).

ARK will not overwrite your custom settings, but it won't show them in the recovery or XMB settings menus, so you kinda have to manage them yourself.
You can leave all ARK's "official" settings disabled and use nothing but custom settings.

I'll leave this open since it will be interesting to add the ability to disable settings on specific games.

DigiOhhh, 2023-02-12:
> Doesn't work for disabling though (I might check this to see how it can be implemented).
> 
> I'll leave this open since it will be interesting to add the ability to disable settings on specific games.
Hey, I once asked this on GBAtemp and you released a testing version with the feature: https://github.com/PSP-Archive/ARK-4/releases/tag/v1648403895
iirc this worked, but if you want I can test it again. I just need to know if I can simply install it by 'downgrading' version 4.19 to the testing one

JoseAaronLopezGarcia, 2023-02-12:
> Hey, I once asked this on GBAtemp and you released a testing version with the feature: https://github.com/PSP-Archive/ARK-4/releases/tag/v1648403895 iirc this worked, but if you want I can test it [...]

I think you're mistaking plugins with settings. I indeed implemented per-game plugin disabling, but not per-game settings disabling. It is possible but will take some work.

Btw this is off-topic, but if you're Italian (I guess so by your name), can you give this translation a review?
https://github.com/PSP-Archive/ARK-4/blob/main/extras/xmbctrl/translations/XMB_IT.TXT
I made it myself with some help from Google Translate and I wanna make sure it feels natural.

DigiOhhh, 2023-02-12:
> I think you're mistaking plugins with settings. I indeed implemented per-game plugin disabling, but not per-game settings disabling. It is possible but will take some work.
>
> Btw this is off-topic, [...]

Oh yes, you are absolutely correct: I was talking about plugins. Glad to know in the end you implemented it. Looking forward to this other feature then, no rush obviously.

About the translation, I'm happy to help. Should I go with a pull request? Or do you prefer I open a small issue with the suggested corrections?

JoseAaronLopezGarcia, 2023-02-12:
> Oh yes, you are absolutely correct: I was talking about plugins. Glad to know in the end you implemented it. Looking forward to this other feature then, no rush obviously.
>
> About the translation, [...]

A pull request is better I think.

JoseAaronLopezGarcia, 2023-03-03:
Note: as of latest release, you can now enable plugins AND settings on multiple games at once:
GameID1 GameID2 GameID3, overclock, on

Still can't disable settings per game, but should be easy to implement that feature once I'm done with the next big release.

JoseAaronLopezGarcia, 2023-05-26:
New version `4.20.50` now has the ability to disable settings per game.

closed 2023-05-26 by JoseAaronLopezGarcia