PSP-Archive/ARK-4 #80: ARK crashes often in the PS Vita/PS TV's boot and often when running PSP games when it doesn't have the former issue

GitHubCFW, 2023-02-12:
After running ARK, it often crashes in the PS Vita/PS TV's boot and often when running PSP games when it doesn't have the former issue.

Regarding the former problem, it states it will close in 10 seconds and if I don't manually close it, ARK crashes again with the C1-2858-3 error.

Regarding the latter problem, when I'm able to boot ARK in the PS Vita/PS TV, it often crashes with the C1-2858-3 error when running PSP games.

I'm on HENkaku 3.60.

[image]

[image]

JoseAaronLopezGarcia, 2023-02-12:
Closed for being spam. Also known issue with known fix.

closed 2023-02-12 by JoseAaronLopezGarcia

Darthagnon, 2023-02-13:
> Closed for being spam. Also known issue with known fix.

Googling around, and came across this issue. I've had the same one, any chance you could link me to the fix?

JoseAaronLopezGarcia, 2023-02-13:
> > Closed for being spam. Also known issue with known fix.
> 
> Googling around, and came across this issue. I've had the same one, any chance you could link me to the fix?

It's in the readme, a simple reboot of the console fixes the issue.