PSP-Archive/ARK-4 #85: Is there a way to USB connect without exiting from custom launcher?

lorenzofelletti, 2023-02-17:
## ARK-4 Bugs/Issues

#### PSP Model
- [ ] PSP-1000 (Fat)
 
- [ ] PSP-2000 (Slim)

- [X] PSP-3000 (Slim & Bright)

- [ ] PSP-N1000 (PSP Go)

- [ ] PSP-E1000 (Street)

- [ ] PCH-1000/1100 (Vita Fat)

- [ ] PCH-2000 (Vita Slim)

- [ ] VTE-1000 (Vita TV)

#### Ark Version

- 4.19.10.

#### Steps to Reproduce
1. Have autoboot custom launcher selected as "always"
2. Connect to pc via USB
3. Note that the device is not recognized by pc, and I did not find an option to connecting it without disabling autoboot and exiting from custom launcher.

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

JoseAaronLopezGarcia, 2023-02-17:
You can achieve this via plugins, my personal recommendation is Custom Firmware Extender, it can be loaded on XMB, launcher, games and homebrews and it offers a wide variety of useful extras. It also looks very nice and blends in really well with ARK, almost looking like a built-in feature.

lorenzofelletti, 2023-02-17:
I will for sure try it, thank you very much!

closed 2023-02-17 by JoseAaronLopezGarcia