PSP-Archive/ARK-4 #89: Crash when returning from sleep mode in Peggle [NPUG30040]

TheSubPlayer, 2023-02-23:
## ARK-4 Bugs/Issues

#### PSP Model

- [X] PSP-3000 (Slim & Bright)

#### Ark Version

4.19.15 (also tested on 4.19.13)

#### Homebrew/GAME (if applicable)

Peggle [NPUG-30040]

#### Plugins ( if applicable )

No plugin activated in game

#### Steps to Reproduce
1. Launch Game
2. In the menu or in game, put the PSP in sleep mode
3. The game will crash immediately or after a few seconds

#### Recovery settings

The error will continue no matter what settings are active

JoseAaronLopezGarcia, 2023-03-01:
Which was the last version of ARK where it was working fine? If there's any.

JoseAaronLopezGarcia, 2023-03-02:
Fixed: https://github.com/PSP-Archive/ARK-4/releases/tag/r41918

closed 2023-03-02 by JoseAaronLopezGarcia