PSP-Archive/ARK-4 #92: Settings in VSH menu don't stay set (UMD ISO Mode Specifically) (Rock Band Unplugged will not start)

tigglesbruh, 2023-03-01:
## ARK-4 Bugs/Issues

#### PSP Model

- [ ] PSP-3000 (Slim & Bright)

#### Ark Version

- 4.19.18

#### Homebrew/GAME (if applicable)

- Rock Band Unplugged (ULUS-10418)

#### Plugins ( if applicable )

 - Category lite, CDDA Enabler, NPLOADER

#### Steps to Reproduce
1. Open VSH menu on XMB
2. Change any setting (none will stay set after resetting VSH)
3. Reset VSH
4. Open VSH menu again, the setting will be default again

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

JoseAaronLopezGarcia, 2023-03-01:
UMD ISO Mode is purely for information purposes, you don't actually get to change the ISO Driver, the one used depends on the format.

tigglesbruh, 2023-03-01:
> UMD ISO Mode is purely for information purposes, you don't actually get to change the ISO Driver, the one used depends on the format.

But I can change it on PRO CFW. In fact, I reinstalled PRO and have no trouble with this at all. The only reason I tried ARK was because of the UMD Video region switch, which works great, but then this problem came up. Is there any way you guys could just implement the UMD video region change as a plugin for PRO? Or fix whatever problem there is with the UMD ISO mode?

JoseAaronLopezGarcia, 2023-03-01:
> But I can change it on PRO CFW. In fact, I reinstalled PRO and have no trouble with this at all. The only reason I tried ARK was because of the UMD Video region switch, which works great, but then [...]

How is this a problem? Inferno 2 ISO Driver is far superior to any old ISO driver in PRO, why would you want to change it and go back to an older, slower, less compatible one?.

tigglesbruh, 2023-03-01:
> How is this a problem? Inferno 2 ISO Driver is far superior to any old ISO driver in PRO, why would you want to change it and go back to an older, slower, less compatible one?.

Because for whatever reason, Rock Band doesn't work on Inferno 2, but it works on NP9660. The option would be nice.

JoseAaronLopezGarcia, 2023-03-01:
> Because for whatever reason, Rock Band doesn't work on Inferno 2, but it works on NP9660. The option would be nice.

In that case you can convert it to PBP format so it always uses NP9660 on any CFW. Does Rock Band not work with the orignal Inferno or M33 drivers?

tigglesbruh, 2023-03-01:
> In that case you can convert it to PBP format so it always uses NP9660 on any CFW. Does Rock Band not work with the orignal Inferno or M33 drivers?

How can I convert to PBP? And as far as I can tell, no.

tigglesbruh, 2023-03-01:
> There's some tutorials on how to convert an ISO to a PBP that will work even on OFW. As for getting the ISO itself to work, it'll probably require some anti-DRM patches, I'll see what I can do.

Awesome, thank you.

closed 2023-04-25 by JoseAaronLopezGarcia

reukiodo, 2024-04-16:
I think this may still be an issue, at least on the PSP Go. I installed 6.61 from Chronoswitch, then installed Infinity2 and Ark4 into firmware and then deleted all the files off the memory card. So now Settings > System Settings > System Information shows '6.61 ARK-4 Infinity∞' so all seems good.

However, trying to set the colors in the VSH MENU > ADVANCED VSH MENU > BG COLORS > Light Black and/or FG COLORS > Red doesn't seem to stick. I've tried setting, rebooting, resetting, power off/on, resetting, GO BACK > EXIT, rebooting, resetting, etc but it always goes back to random colors. :(