PSP-Archive/ARK-4 #94: After exiting games the storage memory disappeared

skelletR, 2023-03-02:
## ARK-4 Bugs/Issues

#### PSP Model
- [ ] PSP-N1000 (PSP Go) firmware 6.61

#### Ark Version

- Version 4.19.18

#### Homebrew/GAME (if applicable)

- All games

#### Plugins ( if applicable )

 - Please list all plugins enabled here.

#### Steps to Reproduce
1. Start a game 
2.press home button and exit
3. and the storage memory disappeared 
4. After reboot is back

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

Pik-a-boo, 2023-04-21:
If its a homebrew the system storage will remain intact after exiting it.

JoseAaronLopezGarcia, 2023-04-21:
Have you tried different combinations of "Redirect ms0 to ef0" using the custom launcher?

skelletR, 2023-04-21:
> If its a homebrew the system storage will remain intact after exiting it. EDIT : It seems like exiting games suddenly wont cause the system storage to disappear. It started after using the SGDeemer plugin and Apollo PSP homebrew tool for some reason.

With all my plugins disable, still have the same problem with new version 4.20

Pik-a-boo, 2023-04-21:
> Have you tried different combinations of "Redirect ms0 to ef0" using the custom launcher?

I only tried from XMB for now with the Old Plugins on PSP Go option enabled.
Now I tried using another homebrew and again the system storage remains intact so I guess homebrew are excluded from this (and can even restore the system storage back).

JoseAaronLopezGarcia, 2023-04-21:
Does the same happen with internal memory? Are the ISOs also being loaded from memory stick?

skelletR, 2023-04-21:
> Have you tried different combinations of "Redirect ms0 to ef0" using the custom launcher?

I try it now with version 4.20 i open the game but when quit the game the psp restart with sony logo snd after some the only the theme of psp just stuck

Pik-a-boo, 2023-04-21:
> Does the same happen with internal memory? Are the ISOs also being loaded from memory stick?

It happens from both internal memory and memory stick when you load ISOs

skelletR, 2023-04-21:
> > Does the same happen with internal memory? Are the ISOs also being loaded from memory stick?
> 
> It happens from both internal memory and memory stick when you load ISOs

Yeah, i try it, the same

krazynez, 2023-04-21:
> If its a homebrew the system storage will remain intact after exiting it. EDIT : It seems like exiting games suddenly wont cause the system storage to disappear. It started after using the SGDeemer plugin and Apollo PSP homebrew tool for some reason.

For apollo homebrew they are working on a fix [Known issue](https://github.com/bucanero/apollo-psp/issues/4), it has hardcoded seplugins that ARK does not use. Not aware of SGDeemer if that does it or not.

skelletR, 2023-04-21:
I have try lot settings, without plugins, but without success, same disappeared, with internal memory and memory card, all the same, something changes after version 4.19.17, this is the last working version for me

JoseAaronLopezGarcia, 2023-04-23:
Fixed: https://github.com/PSP-Archive/ARK-4/releases/tag/r4201

closed 2023-04-23 by JoseAaronLopezGarcia