PSP-Archive/ARK-4 #95: Remotejoylite POPS issue

TheSubPlayer, 2023-03-05:
## ARK-4 Bugs/Issues

- RemoteJoylite doesn't work on POPS if the plugin is set to "all", but it works normally if it's only on "pops"

#### PSP Model

- [x] PSP-3000 (Slim & Bright)

#### Ark Version

- 4.19.18

#### Plugins ( if applicable )

[PLUGINS.TXT](https://github.com/PSP-Archive/ARK-4/files/10893023/PLUGINS.TXT)

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Always

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Always

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Homebrew

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Always

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Always

	</details>

JoseAaronLopezGarcia, 2023-09-12:
Is this still an issue in latest version?

Yeet90, 2024-02-12:
> Is this still an issue in latest version?

Working fine for me on the latest release, the only thing I'm not sure about is you need to restart remotejoylite on the computer side when going from xmb to a game.
[image]
[image]

krazynez, 2024-02-12:
> > Is this still an issue in latest version?
> 
> Working fine for me on the latest release, the only thing I'm not sure about is you need to restart remotejoylite on the computer side when going from xmb to a game. ![Screenshot 2024-02-12 080135](https://private-user-images.githubusercontent.com/126437204/304148046-282d3399-4ec1-40b9-ba16-e87f29b9bfce.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc3NzEwMTAsIm5iZiI6MTcwNzc3MDcxMCwicGF0aCI6Ii8xMjY0MzcyMDQvMzA0MTQ4MDQ2LTI4MmQzMzk5LTRlYzEtNDBiOS1iYTE2LWU4N2YyOWI5YmZjZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxMlQyMDQ1MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZTJhNTVhZDQwNjVkMDNlYmRmZTAzMDE5ZGZhOTAyZGRjZWIyNjJlMzJiYzllNjJjMzk3N2MxYzE2ODI2MWZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.lEuU3uVAfQK_kSmrFFLVdSDC1kwu8hEjaeXyqR8F4ZY) ![Screenshot 2024-02-12 080304](https://private-user-images.githubusercontent.com/126437204/304148054-87ded031-abd2-41a2-be8a-c0227a3cbd92.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc3NzEwMTAsIm5iZiI6MTcwNzc3MDcxMCwicGF0aCI6Ii8xMjY0MzcyMDQvMzA0MTQ4MDU0LTg3ZGVkMDMxLWFiZDItNDFhMi1iZThhLWMwMjI3YTNjYmQ5Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxMlQyMDQ1MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xY2VmYmQ1NDk3YzljOTg2NzM5NmIwY2ZmYTdlNDU4NGZkZTYyMmNlY2Q2MGQwOWMwNGU1ODkwNzQ5MDVlOTg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UKoxqDzw7wwK1Taih__wCiCMfsf1_qeVMM2HZBYmzFM)

wait that isn't a PS1 game though unless I am incorrect.

Yeet90, 2024-02-12:
> wait that isn't a PS1 game though unless I am incorrect.

Nope its not a PS1 game. I just tried FF7 with it and its broken not getting anything. also idk if this is helpful or not but the usb driver it uses doesn't show up in device manager once a PS1 game is started. unplugging and re-plugging it doesn't help.

krazynez, 2024-02-12:
> Nope its not a PS1 game. I just tried FF7 with it and its broken not getting anything. also idk if this is helpful or not but the usb driver it uses doesn't show up in device manager once a PS1 game [...]

can you also add the `pops, flash0:/kd/usb.prx, on` (needed for pops with remotejoy) then try again.

Yeet90, 2024-02-12:
> can you also add the `pops, flash0:/kd/usb.prx, on` (needed for pops with remotejoy) then try again.

I added usb plugin and the results are still the same.

JoseAaronLopezGarcia, 2024-02-12:
> > can you also add the `pops, flash0:/kd/usb.prx, on` (needed for pops with remotejoy) then try again.
> 
> I added usb plugin and the results are still the same.

make sure that usb.prx is loaded before the other plugins

Yeet90, 2024-02-12:
> make sure that usb.prx is loaded before the other plugins

That fixed it, working just fine now 
[image]

krazynez, 2024-02-12:
> That fixed it, working just fine now [...]

Thanks for testing gonna close the this then.

JoseAaronLopezGarcia, 2024-02-12:
> Thanks for testing gonna close the this then.

We can't close it cause the original reporter hasn't responded yet. Lets give it some days.

closed 2024-03-09 by JoseAaronLopezGarcia