PSP-Archive/ARK-4 #97: Infinity shutting down PSP when attempting to run on ARK CFW

PixelateVision, 2023-03-09:
## ARK-4 Bugs/Issues

#### PSP Model
- [ ] PSP-1000 (Fat)
 
- [ ] PSP-2000 (Slim)

- [X] PSP-3000 (Slim & Bright)

- [ ] PSP-N1000 (PSP Go)

- [ ] PSP-E1000 (Street)

- [ ] PCH-1000/1100 (Vita Fat)

- [ ] PCH-2000 (Vita Slim)

- [ ] VTE-1000 (Vita TV)

#### Ark Version

- 4.19.18

#### Steps to Reproduce
1. Installed OFW 6.60 (brand new PSP 3000 out of the box)
2. Installed ARK-4 v4.19.18
3. Installed Infinity 2 from OFW
4. Tried loading ARK Live and then running Infinity 2 to configure permanent boot

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

I see that there have been a couple similar issues closed before- but none of them seem to have a solution. 

Every time I try to run the Infinity tool that comes bundled with ARK-4 while ARK Live is running, the PSP goes to a black screen and then shuts itself off. I was able to successfully install it on OFW, but it will not run while ARK Live is loaded - only when OFW is running (which will not allow me to set the permanent CFW).

TheSubPlayer, 2023-03-09:
Install Infinity and then run live loader

PixelateVision, 2023-03-09:
> Install Infinity and then run live loader

That is what I'm doing; I installed Infinity on OFW, then I'm running Live Loader and trying to run Infinity again (to set the permanent ARK). It's when I try running Infinity the second time that it just shows a black screen and shuts the PSP down.

TheSubPlayer, 2023-03-09:
> > Install Infinity and then run live loader
> 
> That is what I'm doing; I installed Infinity on OFW, then I'm running Live Loader and trying to run Infinity again (to set the permanent ARK). It's when I try running Infinity the second time that it just shows a black screen and shuts the PSP down.

Configure Infinity first and install Live Loader only at the end.
Then let me know if it worked

PixelateVision, 2023-03-09:
> Configure Infinity first and install Live Loader only at the end. Then let me know if it worked

I will try that. Apologies; I'm a bit more of a noob; how would I start over from scratch? Do I need to use Chronoswitch to restore it back to stock OFW?

TheSubPlayer, 2023-03-09:
> > > > Instale o Infinity e, em seguida, execute o carregador ao vivo
> > > 
> > > 
> > > É isso que estou fazendo; Eu instalei o Infinity no OFW, então estou executando o Live Loader e tentando executar o Infinity novamente (para definir o ARK permanente). É quando eu tento executar o Infinity pela segunda vez que ele apenas mostra uma tela preta e desliga o PSP.
> > 
> > 
> > Configure o Infinity primeiro e instale o Live Loader somente no final. Então deixe-me saber se funcionou
> 
> Vou tentar isso. Desculpas; Eu sou um pouco mais de um noob; como eu recomeçaria do zero? Preciso usar o Chronoswitch para restaurá-lo de volta ao estoque OFW?

no need, you can try normally

PixelateVision, 2023-03-09:
> no need, you can try normally

That did the trick! Thank you for your help. :)

closed 2023-03-09 by PixelateVision

TheSubPlayer, 2023-03-09:
> That did the trick! Thank you for your help. :)

Glad to help, enjoy this amazing CFW