PSP-Archive/ARK-4 #98: Large list of games crashes ark

KaylaD23, 2023-04-03:
## ARK-4 Bugs/Issues
the psp crashes and turns off when i scroll a large list of games i have 128gb sd card that its pretty much full.
#### PSP Model
- [ ] PSP-1000 (Fat)
 
- [ ] PSP-2000 (Slim)

- [X] PSP-3000 (Slim & Bright)

- [ ] PSP-N1000 (PSP Go)

- [ ] PSP-E1000 (Street)

- [ ] PCH-1000/1100 (Vita Fat)

- [ ] PCH-2000 (Vita Slim)

- [ ] VTE-1000 (Vita TV)

#### Ark Version

4.19.18

#### Homebrew/GAME (if applicable)

#### Plugins ( if applicable )

Categories lite and cxmb

#### Steps to Reproduce
1.Go to games and scroll the list 
2.
3.
4.

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

JoseAaronLopezGarcia, 2023-04-03:
If you're using categories lite then how come you have a big list of games? The whole purpose of categories lite is so there aren't big lists to prevent this sort of crashes.

KaylaD23, 2023-04-03:
Oh forgot to mention i was on lme before which never give me the crash and shutoff issue.

Crimson-foxGITHUB, 2023-04-22:
I have a PSP-1000, can confirm this glitch happens there as well. (hard-crashes the entire PSP) Categories Lite is an XMB plugin, not an ARK plugin.

JoseAaronLopezGarcia, 2023-04-23:
I have an idea why this might be.

JoseAaronLopezGarcia, 2023-04-25:
I still don't understand this issue. The PSP will run out of memory when a lot of games are installed. Other stuff such as plugins will also consume memory. We only have 32MB to work with.
In any case, I've had over 100 games in some memory sticks and never had a crash loading them all, only really really slow load times. This is what Categories Lite was created for, it lets you skip the RAM limitations of the PSP by loading only a few games at a time. In the Custom Launcher you can set it to use the File Browser instead of the Game Browser, which achieved the same as Categories Lite.

In any case: don't have that many games in the same folder. Use Categories Lite and File Browser.

JoseAaronLopezGarcia, 2023-04-29:
This issue may have been fixed in the latest release. I'm closing it for now.

closed 2023-04-29 by JoseAaronLopezGarcia

JoseAaronLopezGarcia, 2023-05-09:
This pre-release should fix this issue:
https://github.com/PSP-Archive/ARK-4/releases/tag/v1683667191