PSP-Archive/ARK-4 #99: FTP client doesn't work properly with anonymous logins, only ip addresses can be used.

acidflare, 2023-04-08:
## ARK-4 Bugs/Issues

#### PSP Model
- [ ] PSP-1000 (Fat)
 
- [X] PSP-2000 (Slim)

- [ ] PSP-3000 (Slim & Bright)

- [ ] PSP-N1000 (PSP Go)

- [ ] PSP-E1000 (Street)

- [ ] PCH-1000/1100 (Vita Fat)

- [ ] PCH-2000 (Vita Slim)

- [ ] VTE-1000 (Vita TV)

#### Ark Version

- Enter Ark Release here (press Select it shows the version at the top of vshmenu). If using beta build put commit version.
4.20
#### Homebrew/GAME (if applicable)

- If using retail game please add GAME ID i.e: UCUS-98713 

#### Plugins ( if applicable )

 - Please list all plugins enabled here.

#### Steps to Reproduce
1.Attempting to use ftp client with an anonymous login
2.only ip address works, using a url brings you back to the memory card in the file manager
3.Once connected to the server only refresh and disconnect items are displayed

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [ ] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH
        - [X] XMB
	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

JoseAaronLopezGarcia, 2023-04-11:
This is more of a feature request than a bug report. The code only supports entering an IP (not even a port lol). I'll see what I can do, but most likely it will require adding some sort of DNS resolving to turn the URL into an IP.

acidflare, 2023-04-11:
While I can see the url portion being a feature request, not being able to connect to a server and view files while being anonymous is a bit of an issue.

JoseAaronLopezGarcia, 2023-04-11:
> While I can see the url portion being a feature request, not being able to connect to a server and view files while being anonymous is a bit of an issue.

Non-anonymous login is also feature request rather than a bug/issue. That one should be easier to implement.

acidflare, 2023-04-11:
Maybe my wording was a little confusing. It's not about non-anonymous logins.. it's about when you are using an anonymous login the server only displays a refresh and disconnect item which are both shown as folders. Or at the very least that was my experience with it.

acidflare, 2023-04-11:
All of this aside myself and the rest of the community would probably enjoy having a working FTP client with the more popular features being implemented, the PSP hasn't had a properly working FTP client since fw 1.50.

Does the current version support PASV and keep-alive commands?

krazynez, 2023-04-11:
> All of this aside myself and the rest of the community would probably enjoy having a working FTP client with the more popular features being implemented, the PSP hasn't had a properly working FTP client since fw 1.50.
> 
> Does the current version support PASV and keep-alive commands?

Passive mode yes [PASV](https://github.com/PSP-Archive/ARK-4/blob/main/extras/menus/arkMenu/src/ftp/ftp.cpp#L92). 

The timeout is being handled from the [NOOP](https://github.com/PSP-Archive/ARK-4/blob/6607ebf1df3af77e1c195872a000f99b72925855/extras/menus/arkMenu/src/ftp/ftp.cpp#L831) to prevent the timeout from server and client.

JoseAaronLopezGarcia, 2023-04-11:
> All of this aside myself and the rest of the community would probably enjoy having a working FTP client with the more popular features being implemented, the PSP hasn't had a properly working FTP client since fw 1.50.
> 
> Does the current version support PASV and keep-alive commands?

Unfortunately the FTP client and server code in ARK is based on the old ones with some minor bug fixes. It would be ideal to actually rewrite them both into something better but that takes time.

JoseAaronLopezGarcia, 2023-04-23:
You can now try this beta to test this new feature: https://github.com/PSP-Archive/ARK-4/releases/tag/v1682269406
Works fine with some servers but not all of them (tested and working with ftp.eiffel.org, but others such as libvirt.org don't work).
Still in development but it's better than before.
You can also now specify a different port (other than default 21) by using the standard address:port notation.

JoseAaronLopezGarcia, 2023-04-25:
This release should allow the FTP client to do what you want. Though most of the servers I've tested either don't work or crash the client after a few browsing.
https://github.com/PSP-Archive/ARK-4/releases/tag/r4202

closed 2023-04-25 by JoseAaronLopezGarcia