PSP-Archive/ARK-4 #100: CDDA Enabler doesn't work correctly on popsloader v3

Crimson-foxGITHUB, 2023-04-20:
## ARK-4 Bugs/Issues

#### PSP Model
- [X] PSP-1000 (Fat)
 
- [ ] PSP-2000 (Slim)

- [ ] PSP-3000 (Slim & Bright)

- [ ] PSP-N1000 (PSP Go)

- [ ] PSP-E1000 (Street)

- [ ] PCH-1000/1100 (Vita Fat)

- [ ] PCH-2000 (Vita Slim)

- [ ] VTE-1000 (Vita TV)

#### Ark Version

- 4.19.18

#### Homebrew/GAME (if applicable)

- Any PSX game that uses CDDA Audio(e.g. Gran Turismo 2)

#### Plugins ( if applicable )

 - POPSLoader v3, CDDA Enabler

#### Steps to Reproduce
1. Enable POPSLoader, Popscore, and CDDA Enabler
2. Launch a PSX game that uses CDDA audio(any Popsloader version will produce this bug, Gran Turismo 2 and Hydro Prop Racing definitely do)
3. Happens both when launching from Custom Loader and XMB

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [ ] Disabled
	- [X] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

JoseAaronLopezGarcia, 2023-04-20:
You're not meant to install popscore as a plugin.

JoseAaronLopezGarcia, 2023-04-21:
Closing this as it doesn't seem to be a bug but a misconfiguration issue.
@Crimson-foxGITHUB feel free to reopen if you feel the issue is not solved.

closed 2023-04-21 by JoseAaronLopezGarcia