PSP-Archive/ARK-4 #101: Feature: Option to hide ARK and custom firmware paths from ISO games

xslendix, 2023-04-21:
Some later ISO games check for paths such as `ms0:/seplugins`, a way to hide those paths would be great, especially considering some games go ahead and DELETE YOUR DATA, such as save files and plugins. Some option to do this per game basis would be great, tsanks. For example, a game that does this is djmax portable 3, thankfully without the deletion part. This game just exists after a while.

JoseAaronLopezGarcia, 2023-04-21:
This is already implemented as part of the Stargate anti-DRM module that runs in ARK when using ISO/CSO/PBP games.
There's just no option to "enable/disable", it's always enabled.

JoseAaronLopezGarcia, 2023-04-23:
Closing this since the feature is already available and always active. If any issues arise from having this patch always applied you can open a new issue thread.

closed 2023-04-23 by JoseAaronLopezGarcia