PSP-Archive/ARK-4 #103: "Error (1) exploit failed" during Ark installer

Mephio, 2023-04-25:
## ARK-4 Bugs/Issues
Ark loader error during installer: Error (1) exploit failed.

#### PSP Model
- [X] PSP-1000 (Fat)
 
- [ ] PSP-2000 (Slim)

- [ ] PSP-3000 (Slim & Bright)

- [ ] PSP-N1000 (PSP Go)

- [ ] PSP-E1000 (Street)

- [ ] PCH-1000/1100 (Vita Fat)

- [ ] PCH-2000 (Vita Slim)

- [ ] VTE-1000 (Vita TV)

#### Ark Version

- 4.20.02

#### Homebrew/GAME (if applicable)

#### Plugins ( if applicable )

 - No plugin installed, firmware stock

#### Steps to Reproduce
1. Installed ARK_01234 folder into /PSP/SAVEDATA/ folder
2. Installed ARK_Live folder into /PSP/GAME/ folder
3. Launched ARK Loader

#### Recovery settings
- USB Charge  
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- OverClock 

	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- PowerSave
	<details>
		<summary>Option type (click to view)</summary>
	
	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Autoboot Launcher

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Disable PSP Go Pause

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Unlock Extra Memory

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Memory Stick Speedup

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

- Inferno Cache 

	<details>
		<summary>Option type (click to view)</summary>

	- [X] Disabled
	- [ ] Always
	- [ ] Game
	- [ ] UMD
	- [ ] Homebrew
	- [ ] Pops
	- [ ] VSH

	</details>

JoseAaronLopezGarcia, 2023-04-25:
Make sure you have installed ARK_01234 in /PSP/SAVEDATA/

krazynez, 2023-04-25:
I cannot reproduce this.

Mephio, 2023-04-25:
Last debug version worked.
I'll try to redownload release version.

JoseAaronLopezGarcia, 2023-04-26:
Sometimes the compiler and/or build scripts would get confused and compile/copy the wrong K.BIN file (the kernel exploit). So sometimes you get the Vita's Kxploit file instead of the PSP's or viceversa. I have no idea what causes this issue as most of the time it works fine.

Anyways it should be working fine on latest release, if not then feel free to re-open this thread.

closed 2023-04-26 by JoseAaronLopezGarcia

JORGETECH, 2023-08-11:
I think this bug is also present in version 4.20.63, I had the same issue when running Ark Loader on a PSP-1000 (Fat), I could fix the issue by downloading the K.BIN from the previous release.

krazynez, 2023-08-12:
> I think this bug is also present in version 4.20.63, I had the same issue when running Ark Loader on a PSP-1000 (Fat), I could fix the issue by downloading the K.BIN from the previous release.

Can you provide the steps you did initially?

JoseAaronLopezGarcia, 2023-08-12:
> I think this bug is also present in version 4.20.63, I had the same issue when running Ark Loader on a PSP-1000 (Fat), I could fix the issue by downloading the K.BIN from the previous release.

I've updated the release with the proper K.BIN files.