PSP-Archive/ARK-4 #106: [REQUEST] Enable/Disable Plugins

TheSubPlayer, 2023-05-02:
### New Feature

This is an existing function in PRO CFW, I thought it had no practical use, until I realized that it is very useful for the user to perform tests and identify causes of possible errors/crashes.

[image]

krazynez, 2023-05-02:
So implementing that would be pretty easy, only thing I can think of however is doing the inverse to get it back to what the user had before hand. (Though I think I know a easy way to do that off the top of my head)

JoseAaronLopezGarcia, 2023-05-02:
There's an extremely simple way to disable all plugins: use the file browser and rename the PLUGINS.TXT file to something else. I don't really see any reason to have a "plugins on/off" setting.

TheSubPlayer, 2023-05-02:
Yes, but that would disable all plugins, rather than just GAME or XMB for example

krazynez, 2023-05-02:
Right initally I was thinking what @JoseAaronLopezGarcia was thinking but per runtime could be helpful for testing. Not so much for end users.

JoseAaronLopezGarcia, 2023-05-02:
Step 1: power on your PSP holding R button to enter `Recovery Menu`.
Step 2: use file browser and locate `/SEPLUGINS/PLUGINS.TXT`.
Step 3: rename the file to something, anything, else.
Step 4: Exit recovery menu and see results.

It can't get any easier than that. I really don't see the need for something like this to be a setting.

JoseAaronLopezGarcia, 2023-05-05:
https://github.com/PSP-Archive/ARK-4/releases/tag/v1683246907

closed 2023-05-05 by JoseAaronLopezGarcia