PSP-Archive/ARK-4 #108: Can't open the Plugins Manager in the XMB

IceShikurai28, 2023-05-04:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.13

### Steps to reproduce

Can't access the Plugins Manager in the XMB.
My PSP freezes then shuts down immediately.

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Always

JoseAaronLopezGarcia, 2023-05-05:
what is the contents of plugins.txt if any?

IceShikurai28, 2023-05-06:
> what is the contents of plugins.txt if any?

None, I figured it only happens when there's no plugin installed at all. I guess I just gotta avoid it now that I don't have any idea which plugins to use or add, lol.

krazynez, 2023-05-07:
By default we have an "example" plugin to prevent that from happening.

closed 2023-05-07 by IceShikurai28

JoseAaronLopezGarcia, 2023-05-07:
Was this fixed?

cyberbasty, 2024-04-29:
what's the example plugin ? i have the same problem

krazynez, 2024-04-29:
> what's the example plugin ? i have the same problem

its just a place holder. You need to have actual plugins in your PLUGINS.TXT to show up.

cyberbasty, 2024-04-29:
ok thanks

cyberbasty, 2024-04-29:
i have to put plugins for vsh or for game ?

cyberbasty, 2024-04-29:
> > i have to put plugins for vsh or for game ?
> 
> Depends on the plugin. Read the instructions of the plugin to know if it's for VSH (XMB), PSP Games or PS1 Games (POPS).

remotejoylite is for all of them

JoseAaronLopezGarcia, 2024-04-29:
> > > i have to put plugins for vsh or for game ?
> > 
> > 
> > Depends on the plugin. Read the instructions of the plugin to know if it's for VSH (XMB), PSP Games or PS1 Games (POPS).
> 
> remotejoylite is for all of them

then you can use all/always