PSP-Archive/ARK-4 #109: Custom Launcher

KarnFu, 2023-05-05:
### New Feature

I love the custom launcher and I want use it more than XMB. So a digital clock and date display next to the battery icon would be cool, also it would be awesome to have more navigation options in the games menu like for example, I want to delete a game or check game size (Press L Button when highlighting a game to open menu for available options).
A better music player that has background play within the launcher, and an improved image viewer (crahes when opening other images that display fine on XMB, I will post an issue about this after I test)
Finally the ability to update ARK from the custom launcher (I have to disable Auto Bootloader and use XMB so I can update).

TheSubPlayer, 2023-05-05:
I don't know if I should open a new topic just for this, but the image viewer doesn't handle images above 480x272 well

https://user-images.githubusercontent.com/110436924/236467587-c5878291-c666-4753-8700-f72949cd75fc.mp4

JoseAaronLopezGarcia, 2023-05-05:
yeah there isn't that much memory available, specially if you have other things in the background like the games list, browser list, theme data, etc.
The only way to fix this would be to offload a lot of things in the launcher and reload them again when exiting from image viewer. This will be slow though, specially having to load the theme again.

JoseAaronLopezGarcia, 2023-05-06:
Latest release implements some of these features:
https://github.com/PSP-Archive/ARK-4/releases/tag/r42015

KarnFu, 2023-05-06:
> Latest release implements some of these features: https://github.com/PSP-Archive/ARK-4/releases/tag/r42015

Looks great I will have a test and get back to you

KarnFu, 2023-05-07:
Bro you are awesome and quickly becoming a legend. Everything looks great and works great. This launcher makes a difference in the modern world. Off - topic, Inferno V2 has really improved (I'm the one from Discord who had an issue with God of War Ghost of Sparta CSO audio popping) The slowdowns I usually encounter in certain games have noticeably disappeared on all formats; CSO, ISO, Eboots even UMD games. I think overall gaming performance is great and stable. Oh one more thing to add in the Custom Launcher's File Browser is info/indicator of how much space is available in the memory stick. This should be useful for when using FTP to transfer files and etc.

Thanks again Bro

JoseAaronLopezGarcia, 2023-05-07:
> Bro you are awesome and quickly becoming a legend. Everything looks great and works great. This launcher makes a difference in the modern world. Off - topic, Inferno V2 has really improved (I'm the [...]

Glad you're liking it. If you can, give this beta build a ride and give me some feedback on how the launcher/recovery control is: https://github.com/PSP-Archive/ARK-4/releases/tag/v1683459489

KarnFu, 2023-05-08:
> 

The controls feel a lot smoother than before, still missing a few things though but this is has blown away my expectations

KarnFu, 2023-05-08:
I don't know weather or not to issue these as bugs but I found a couple of problems.

-The file browser is functioning as intended but there are some issues like opening folders with multiple files (Like my Gameboy ROMs Folder that has more than a Thousand games, or the Picture folder) This sometimes causes the PSP to crash. The file manager should be faster when opening any folders, CM File Manager is a perfect example but the developer hasn't updated the app for a while now. Please have a look https://github.com/joel16/CMFileManager-PSP 

-Music player needs a section in the launcher menu so that it's easier to have access, another issue is when opening a game with music playing does something weird like the PSP is frozen but when I put the PSP into sleep and back it loads into the game.

-Image Section in the launcher for easy access

-How much free space indicator in the file browser (eg: Free Space 250mb)

-Change Theme should be in launcher settings

-Ability to update ARK from launcher settings, I have to disable launcher and go back to XMB. The launcher looks better than XMB so I want all of ARK's settings in there

JoseAaronLopezGarcia, 2023-05-09:
Here's my thoughts on this:

- File Browser crashing on large lists: this one is a no brainer, just consider all the things loaded at once in the Custom Launcher: all your games (ISO, PBPs, etc), an FTP Server that can be running in the background while you still use the file browser, all the background animations, the theme, etc. We only have 24MB to work with, so it's no wonder that a much simpler file browser that only does that, browser files, and nothing else, will work much better than the one built into the launcher. My recommendations here: turn on Force Extra Memory on anything that isn't a PSP 1K, set the Menu Settings to load the File Browser as the main "app" instead of the Game launcher, so it doesn't load games at startup, leaving more RAM available for the file browser.

- Music Player and Image Viewer having their own section: I don't actually like this idea for various reasons. The main one being: what's the difference between the Music Player and the File Browser? In any case these sections would be much more limited than using the file browser. With the browser you can go to one folder, open a bunch of MP3s, go to another folder, open some more, etc. It also makes more sense for me to use the file browser to open files, same as you would on PC (you don't generally open VLC/etc to then use it to open the file, you generally browse to the file using the file explorer and double click it to bring up the player).

- Free space calculation: not a bad idea, I'll give it some thought.

- Theme changer in settings: this adds one inconvinience, sort of, aside from the extra code (that I'd like to avoid cause like I said, 24MB of RAM...), I would have to decide: where do you install the other themes? I would have to hardcode and force users to copy the themes folder to some specific path, rather than being able to copy it anywhere and use the file browser. I wouldn't completely turn down this idea, but I would not give it much priority, there's more important things at the moment.

- Update ARK from launcher: this one is a no brainer, I'll get to it.

KarnFu, 2023-05-09:
> Here's my thoughts on this:
> 
> * File Browser crashing on large lists: this one is obvious, just consider all the things loaded at once in the Custom Launcher: all your games (ISO, PBPs, etc), an FTP Server that can be running in the background while you still use the file browser, all the background animations, the theme, etc. We only have 24MB to work with, so it's no wonder that a much simpler file browser that only does that, browser files, and nothing else, will work much better than the one built into the launcher. My recommendations here: turn on Force Extra Memory on anything that isn't a PSP 1K, set the Menu Settings to load the File Browser as the main "app" instead of the Game launcher, so it doesn't load games at startup, leaving more RAM available for the file browser.
> * Music Player and Image Viewer having their own section: I don't actually like this idea for various reasons. The main one being: what's the difference between the Music Player and the File Browser? In any case these sections would be much more limited than using the file browser. With the browser you can go to one folder, open a bunch of MP3s, go to another folder, open some more, etc. It also makes more sense for me to use the file browser to open files, same as you would on PC (you don't generally open VLC/etc to then use it to open the file, you generally browse to the file using the file explorer and double click it to bring up the player). The other reason being that, as I said, we don't have much RAM, so we need to keep the code as simple as possible, this would just add extra bloat.
> * Free space calculation: not a bad idea, I'll give it some thought.
> * Theme changer in settings: this adds one inconvinience, sort of, aside from the extra code (that I'd like to avoid cause like I said, 24MB of RAM...), I would have to decide: where do you install the other themes? I would have to hardcode and force users to copy the themes folder to some specific path, rather than being able to copy it anywhere and use the file browser. I wouldn't completely turn down this idea, but I would not give it much priority, there's more important things at the moment.
> * Update ARK from launcher: this one is a no brainer, I'll get to it.

Awesome reply, I just learned more about the system's limitations. But either way I am very happy with ARK and how far it got. Someday soon I will buy y'all a 🍺

closed 2023-05-13 by KarnFu