PSP-Archive/ARK-4 #110: 4.20.15 not available through system update?

Dark98, 2023-05-06:
Is this deliberate?

krazynez, 2023-05-07:
I need more details. We (main devs) have had no issues. However there was one occurrence that was sorted out with the psp-updatelist.txt file that was resolved pre 4.20.15.

Can you show what error occurs (screenshot/pictures) if so, specific time frame (in UTC) so I can see the logs to see if its something server side.

Dark98, 2023-05-07:
I'm using ark 4.20.14, Updates through system updater worked to get to this release(I updated at the time 4.20.14 was the latest)

01:56-01:57 UTC Time

[image]

krazynez, 2023-05-07:
I am not seeing anything for that time frame.

We have 4.20.16 out now. You can try that and see.

If you can, join the [PSP Hombrew Discord](https://discord.com/invite/bePrj9W) and we can do proper timing to see if I still can't see you try to connect to the server.

Dark98, 2023-05-07:
This was a self made error!..
If anyone has this problem in the future, make sure VSH Region Change is set to "Default"

closed 2023-05-07 by Dark98