PSP-Archive/ARK-4 #111: Load failed. The data is corrupted

skelletR, 2023-05-08:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.17

### Homebrew/GAME (if applicable)

All

### Steps to reproduce

Open a game and try to load a save

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

skelletR, 2023-05-08:
> ### PSP/Vita Model
> PSP-N1000 (PSP Go)
> 
> ### ARK Version
> 4.20.17
> 
> ### Homebrew/GAME (if applicable)
> Some games with autosave feature
> 
> ### Plugins ( if applicable )
> _No response_
> 
> ### Steps to reproduce
> Open a game and try to load a save
> 
> ### USB Charge
> Always
> 
> ### Overclock
> Disabled
> 
> ### Power Save
> Disabled
> 
> ### Autoboot Launcher
> Disabled
> 
> ### Disable PSP Go Pause
> Disabled
> 
> ### Unlock Extra Memory
> Disabled
> 
> ### Memory Stick Speedup
> Disabled
> 
> ### Inferno Cache
> Disabled

Fix it with MagicSave Plugins

closed 2023-05-08 by skelletR

JoseAaronLopezGarcia, 2023-05-08:
This is a regression in 4.20.17 due to the new "old homebrew" patch. I've improved it to get rid of this issue in the latest beta.

JoseAaronLopezGarcia, 2023-05-08:
This release fixes it: https://github.com/PSP-Archive/ARK-4/releases/tag/r42018

skelletR, 2023-05-08:
> This release fixes it: https://github.com/PSP-Archive/ARK-4/releases/tag/r42018

Nice. Now i have a problem with infinity boot in the game Jak and Daxter: The Lost Frontier until now, when show the logo sony after show a meteorite fell and black screen and again from the metereon scene infinity