PSP-Archive/ARK-4 #114: custom launcher stuck on psp3000 03g board

ca1e, 2023-05-11:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.18

### Steps to reproduce

1. run ark launcher
2. infinity it, reboot
3. open vsh menu,select custom launcher
4. custom launcher loading game icons
5. stuck here

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

krazynez, 2023-05-11:
Please update to 4.20.20 and verify if the issue is still happening.

ca1e, 2023-05-11:
> Please update to 4.20.20 and verify if the issue is still happening.

updated and it still happen

JoseAaronLopezGarcia, 2023-05-11:
Could be one of the games, or homebrew, causing issues. Perhaps due to non-latin characters in file names.

ca1e, 2023-05-12:
> Could be one of the games, or homebrew, causing issues. Perhaps due to non-latin characters in file names.

any debug info can I provide?

ca1e, 2023-05-13:
@JoseAaronLopezGarcia  hi I just have tested that there IS some non-latin characters in file names that crash the launcher.I use some chinese to identify what game it is.so is there some ways to fix that? like use utf8?

JoseAaronLopezGarcia, 2023-06-18:
Can you please try the latest pre-release?

ca1e, 2023-06-19:
> Can you please try the latest pre-release?

Ok, I had been tested it and the latest pre-release does solved some issues.the problem is the none-latin character doesn't show in the top bar when game has been seleted, and the launcher still stuck sometimes when take a while.

JoseAaronLopezGarcia, 2023-06-19:
> > Can you please try the latest pre-release?
> 
> Ok, I had been tested it and the latest pre-release does solved some issues.the problem is the none-latin character doesn't show in the top bar when game has been seleted, and the launcher still stuck sometimes when take a while.

I can't get non-latin characters to show up for file names, it'll show the shortname instead.

JoseAaronLopezGarcia, 2023-07-06:
Closing as solved.

closed 2023-07-06 by JoseAaronLopezGarcia

MasterTJ123, 2024-10-14:
I'm having the same problem with my PSP 3000 version 6.61 using ARK-4 with cIPL. Unfortunately, I activated the option to boot directly into the custom CFW menu, and now my PSP can't start; it goes into the menu, freezes, and shuts down. What should I do? How can I recover it?

krazynez, 2024-10-14:
> I'm having the same problem with my PSP 3000 version 6.61 using ARK-4 with cIPL. Unfortunately, I activated the option to boot directly into the custom CFW menu, and now my PSP can't start; it goes into the menu, freezes, and shuts down. What should I do? How can I recover it?

Hold select on boot.

MasterTJ123, 2024-10-14:
> > I'm having the same problem with my PSP 3000 version 6.61 using ARK-4 with cIPL. Unfortunately, I activated the option to boot directly into the custom CFW menu, and now my PSP can't start; it goes into the menu, freezes, and shuts down. What should I do? How can I recover it?
> 
> Hold select on boot.

Sorry, I know that was a beginner's mistake, lol. Thank you for the quick and accurate response; I managed to revert it. That said, it is a bug that needs to be resolved, right? I believe the issue is due to the Visual Novel ISOs I have on the SD card, some being in Japanese and others translated into English.

krazynez, 2024-10-14:
Hmm possibly could be an issue still. I can retest some JP games tomorrow and see. Titles would be welcome if possible.

MasterTJ123, 2024-10-14:
> Hmm possibly could be an issue still. I can retest some JP games tomorrow and see. Titles would be welcome if possible.

- Ano hi Mita Hana no Namae wo Bokutachi wa Mada Shiranai (Japanese version)
- Boku wa Tomodachi ga Sukunai (Japanese version)
- Final Fantasy (English version)
- Mirai Nikki - 13 Hitome no Nikki Shoyuusha Re-Write (Limited Edition) (Japanese version)
- Sword Art Online - Infinity Moment (English translation)
- To Love-Ru - Doki Doki Rinkai Gakkou Hen (Japanese version)
- Toradora! Portable (English translation)

MasterTJ123, 2024-10-20:
> Hmm possibly could be an issue still. I can retest some JP games tomorrow and see. Titles would be welcome if possible.

I did some tests and realized that the problem is only with the ISO of the game 'Sword Art Online - Infinity Moment,' more specifically version 2.0 of the fan-made translation. Additionally, for some reason, neither the image nor the preview music for game selection work in the original PSP menu, even though the files are present inside the ISO. Can you take a look? I'm not sure to what extent this is something you can fix or if it should be resolved by the creator of the translation. The translation was done by "HoaiTrung97 Fan Translation."