PSP-Archive/ARK-4 #115: add translation for chinese

ca1e, 2023-05-11:
### New Feature

place the _cht.txt but dont work

```
CFW设置
插件管理
USB充电
超频
省电
自启动加载器
PSP Go禁止暂停
强制额外内存
Memory Stick超频
Inferno缓存
PSP Go旧插件支持
跳过启动动画
隐藏PIC0和 PIC1
阻止休眠PSP Go
隐藏MAC地址
隐藏DLC
关闭LED
```

krazynez, 2023-05-11:
From the looks of things it looks like we are missing Chinese Traditional and Chinese Simplified translates for the XMB, vsh menu you would put in seplugins as satelite_cht.txt that was ported over by PRO's vsh menu. Would be ideal to just have this hook into the xmb's less user work in the end I can work on.

For example for (a)vshmenu: 

`g_messages = (const char**)apply_language("satelite_cht.txt");`

``` 
static char ** apply_language(char *translate_file)
{
...
    sprintf(path, "ms0:/seplugins/%s", translate_file);
    ret = load_translate_table(&message, path, MSG_END);
...
    sprintf(path, "ef0:/seplugins/%s", translate_file);
    ret = load_translate_table(&message, path, MSG_END);
...
}
 ```

JoseAaronLopezGarcia, 2023-05-11:
It should be named XMB_CH1.TXT (for simplified chinese) and XMB_CH2.TXT (for traditional).
I'll add it myself for later releases.

ca1e, 2023-05-12:
> @ca1e If you can, can you provide use with the simplified Chinese translation so we can add that as well. 😃 I would be very much appropriated if so.

Sure, I'd love to do that!

> It should be named XMB_CH1.TXT (for simplified chinese) and XMB_CH2.TXT (for traditional). I'll add it myself for later releases.

actually it should be called `CHS` (for simplified chinese) and `CHT` (for traditional) as official as:). 1 or 2 will confuse the user

JoseAaronLopezGarcia, 2023-05-12:
> > @ca1e If you can, can you provide use with the simplified Chinese translation so we can add that as well. smiley I would be very much appropriated if so.
> 
> Sure, I'd love to do that!
> 
> > It should be named XMB_CH1.TXT (for simplified chinese) and XMB_CH2.TXT (for traditional). I'll add it myself for later releases.
> 
> actually it should be called `CHS` (for simplified chinese) and `CHT` (for traditional) as official as:). 1 or 2 will confuse the user

The problem is I have no idea which one is one, is CH1=CHT or CHS? I have no idea, can you confirm which one is which and I'll do the change?

ca1e, 2023-05-12:
> 
> The problem is I have no idea which one is one, is CH1=CHT or CHS? I have no idea, can you confirm which one is which and I'll do the change?

@JoseAaronLopezGarcia I confirmed CH1 = CHT and of course CH2 = CHS. and what I post above is for CHS

krazynez, 2023-05-12:
Now I am confused why was the file named `_cht.txt` if its the simplified Chinese? Wouldn't it have been named _chs.txt instead?

ca1e, 2023-05-12:
> Now I am confused why was the file named `_cht.txt` if its the simplified Chinese? Wouldn't it have been named _chs.txt instead?

Its a typo Im sorry~

JoseAaronLopezGarcia, 2023-05-12:
This isn't fully completed, we still need the traditional translation, can you provide those?

ca1e, 2023-05-12:
> This isn't fully completed, we still need the traditional translation, can you provide those?

of course here you are
```
CFW設定
挿件管理
USB充電
超頻
省電
自啟動加載器
PSP Go禁止暫停
強制額外記憶體
Memory Stick超頻
Inferno緩存
PSP Go舊挿件支持
跳過啟動動畫
隱藏PIC0和PIC1
封锁休眠PSP Go
隱藏MAC地址
隱藏DLC
關閉LED
```

JoseAaronLopezGarcia, 2023-05-12:
Added, Thanks a lot!

JoseAaronLopezGarcia, 2023-06-23:
hey @ca1e can you also help us translate the custom launcher to chinese?
https://github.com/PSP-Archive/ARK-4/issues/149